DLR-SC/tigl #619: tiglWingGetProfileName not returning error

aidanjungo, 2019-08-21:
Similarly to the [TIXI issue #129](https://github.com/DLR-SC/tixi/issues/129) raised by @aarondettmann, the function  'tiglWingGetProfileName' return an empty string when it cannot find the name of the profile. I am using the Python interface and tigl3 3.0.0rc1 (installed via Anaconda)

joergbrech, 2019-12-12:
I am sorry for the late reply. In which situation does this problem occur exactly?

 - `tiglWingGetProfileName` throws an error, if any of the input arguments is invalid, e.g. if no wing of the specified index exists.
   ```bash
   ERR 12/12 17:16:59 tigl.cpp:1520] Invalid index in CCPACSWings::GetWing
   ```
 - TiGL throws an error if the required xpath `name` is missing from the `wingAirfoil`
   ```bash
   ERR 12/12 17:17:54 src_generated.cpp:22073] Required element name is missing at xpath /cpacs/vehicles/profiles/wingAirfoils/wingAirfoil[1]
   ```
 - TIGL throws a warning if the xpath `name` exists, but is empty:
   ```bash
   WRN 12/12 17:15:22 src_generated.cpp:22069] Required element name is empty at xpath /cpacs/vehicles/profiles/wingAirfoils/wingAirfoil[1]
   ```

aidanjungo, 2019-12-12:
Indeed, I cannot reproduce the error now. However, I did not use the Tigl library alone anymore but the one included in cpacscreator, so maybe the problem has been solved... or do the function "tiglWingGetProfileName" was base on the tixi function "getDoubleElement"...?

joergbrech, 2019-12-19:
I just checked, it seems to work with the *latest* cpacscreator release. The function does not use `getDoubleElement`, and even if it did, the bug fix for TiXI currently only effects the development version and not an official release.

I will close this issue, feel free to reopen it, if you can reproduce it again.

closed 2019-12-19 by joergbrech