DLR-SC/tigl #626: Adding sections with guide curves not working

rainman110, 2019-09-26:
This applies for fuselages and wings

svengoldberg, 2024-11-26:
@joergbrech @merakulix @AntonReiswich We discussed this in the last session. IIRC, our idea was to throw an exception when sections are added in segments with gc. We also agreed on leaving it like this and see whether users are really seeking for this feature.
In this case we could think about implementing way adding those sections. Nevertheless, there are open questions on how exactly this should behave.

If you are fine with my suggested solution (so just throw an error and do not allow it; see https://github.com/DLR-SC/tigl/commit/8f1c05afbf0da9d72ba1640b83ee0abc2a4f1a7a), I can open the PR.

joergbrech, 2024-11-26:
A quick code-review on your commit here:  Lets be specific in the error message that it is not supported out of principle, but it is not supported for now. We could even include the information i the error message that guide curves should be added once all sections are defined, which IMHO makes sense, because that's basically the data dependency in CPACS *(guide curves reference existing sections, not vice versa)*. What do you think?

svengoldberg, 2024-11-26:
Yes of course, that sounds reasonable.
The hint regarding the data dependency is also a good idea. I will add this

PR #1039 (2024-11-29): CPACS-Creator: Add error handling when a section should be added in (or next to) a segment containing guide curves
Currently, in the CPACS-Creator a section can be added within or next to a segment that contains guide curves (issue #626).
Right now, this results in an error occuring elsewhere or undefined behaviour. Therefore, for the moment it should not be allowed for a user to do this.
From a CPACS perspective it seems reasonable: The guide curves are generally defined after all sections are created as the guide curves depent on them.

Fixes #626 

If users seek for this feature, we can thi [... cut]
closed 2024-12-03 by svengoldberg