DLR-SC/tigl #627: Adding a wing section results in wrong wing structure

rainman110, 2019-09-27:


joergbrech, 2024-11-12:
To Do: Reproduce.

svengoldberg, 2024-11-25:
I tried to reproduce the behaviour with the file `simpletest-fuselagekinks.cpacs.xml` from the unittests; a simple wing containing 3 sections and 2 segments.

Here, adding a wing section between the first and second one really leads to a strange result. The adding itself works perfectly fine. The new section (and all others) can also be seen when selected in the CPACS-creator interface.
But for some reason, the last segment, which should actually not be touched in the first place, is not displayed anymore. At least that gives a wrong wing structure (although I cannot tell if that is in the meaning of this specific issue). In addition to that, the missing segment cannot not be drawn when selected explicitely via `Draw->Aircraft->Draw any Component`.

The good (while somehow strange) news: This is not the case if the CPACS file is stored and reloaded. In this case, the wing is created as expected right from the start. Then, all segments again can be drawn again via the explicit call.

Further information: This only happens for the section between the first and second one. If any other section is added, it works right from the start.

Also, it might be sensible to correct the order of the listed sections in the creator interface. Right now, the new section is just put at the end, no matter where it is sorted in.