DLR-SC/tigl #639: Computation of reference area

MarAlder, 2019-10-10:
I'm tracking down a potential bug in the computation of the aspect ratio. The span accounts for the wing+symmetric_wing. However, the reference area (it seems that the symmetry plane is not set correctly, but I'll try to fix it after the present issue) is only half of the wing (neglecting the symmetric part). In `GetReferenceArea` ([line 439](https://github.com/DLR-SC/tigl/blob/master/src/wing/CCPACSWing.cpp)) we only loop over the two segments of the given wing (`m_segments.GetSegmentCount()` returns 2 instead of 4). 

I think `wingGetReferenceArea_success` should test to be close to 3.5  ([line 310](https://github.com/DLR-SC/tigl/blob/master/tests/unittests/tiglWing.cpp)) for this test case, or am I wrong?

What is the best TiGL strategy to handle the symmetric part of the wing?

rainman110, 2019-10-10:
I'll turn the question around:

What would an aerospace engineer expect? How is a reference area in the symmetric case handled? Do you mean the simpletest case?

Currently, the area is only computed for the one side of the wing! Thus it should return approx 1.75. So again, should the ref area be doubled? If so, this would mean a big API change behaviour.

MarAlder, 2019-10-10:
This question was running through my head as well. I fear this leads back to the question how we want to interpret the symmetry attribute in future. Looking forward for the corresponding CPACS discussions..

But the problem in this context: How do we treat the wing span? We need a common approach here to get the correct aspect ratio. But we can open a separate issue for this discussion.

MarAlder, 2019-10-11:
Closed because no bug. Symmetry strategy must be discussed with community.

closed 2019-10-11 by MarAlder