DLR-SC/tigl #657: Conda package with OCC 7.4

trungleduc, 2020-02-24:
Hello, i tried to install tigl from conda with `conda install tigl3 -c dlr-sc` , il works well but this version still uses OCE instead of OCCT, is there any solution to using all internal python API  with OCC 7.4?

Thanks.

rainman110, 2020-02-24:
We don't fully support occt 7.4 yet as there are some tests failing. Therefore, Tigl is still tied to a old occt / oce version.

If we have managed to sort out all problems with the new occt, the python bindings will pull in pythonocc 7.x as well.

trungleduc, 2020-02-25:
Thank you for your quick answer, do you have any guide about compiling Tigl with OCCT? I want to try it while waiting for the official release, the guide on your wiki page is for compiling with OCE.

rainman110, 2020-02-28:
Do you simply want to compile the C++ code or do you need the python extension as well?

The C++ stuff is relatively easy, the python bindings more complicated. I never built the python bindings for OCCT 7.4.0, so problems can be expected.

trungleduc, 2020-03-04:
I need the python extension to use TIGL with another lib which depends on OCCT 7.4

rainman110, 2020-03-04:
Have a look at our windows builds, how the python bindings are created:
https://github.com/DLR-SC/tigl/blob/master/appveyor.yml

The best is, to install all dependencies into a conda environment and let cmake search in that directory. Also, cmake must be called with the argument:
```-DTIGL_BINDINGS_PYTHON_INTERNAL=ON ```

Expect problems during build, as I never build the python bindings with pythonocc 7.4.0!

closed 2020-05-12 by rainman110