DLR-SC/tigl #658: Missing function in Java binding

agodemar, 2020-03-03:
Hi guys, 
I am updating my Java library JPAD in order to import CPACS files using TiGL v3.
When I try the current Java binding provided by `TiglNativeInterface.java` I get a runtime error saying:

`Error looking up function 'tiglExportMeshedFuselageVTKSimpleByUID': Impossible to find the specified procedure.`

It looks like `tiglExportMeshedFuselageVTKSimpleByUID` is not defined in `TiglNativeInterface.java`. In fact, this function is only available in the Fortran binding in `tigl/bindings/fortran/tigl_fortran.cpp`.

I think that adding a new function `tiglExportMeshedFuselageVTKSimpleByUID` in `tigl/src/api/tigl.cpp` declared as `TIGL_COMMON_EXPORT` and declaring a new `public static native int tiglExportMeshedFuselageVTKSimpleByUID` in `tigl/bindings/java/src/de/dlr/sc/tigl3/TiglNativeInterface.java` will address the problem.

rainman110, 2020-03-03:
Hi Agostino,

nice to hear from you. 

The java bindings are unfortunately not created automatically and we might forget to adapt it to api changes sometimes. We'll fix this.

Please note, that there are now also new unified export functions, which you could use as well:
 - [tiglExportComponent](https://dlr-sc.github.io/tigl/doc/latest/group__Export.html#ga0183a3551bd7dc408443204ae2b55db3)
 - [tiglExportConfiguration](https://dlr-sc.github.io/tigl/doc/latest/group__Export.html#ga3af6347dd7d28dcafa5c4d0ed8f80c77)
 - [tiglSetExportOptions](https://dlr-sc.github.io/tigl/doc/latest/group__Export.html#ga22af55197aae3c30ef743b5c7894f71f) 

Depending on the file name ending, the appropriate exporter is used.

agodemar, 2020-03-03:
Hi Martin,
thank you for your prompt reply. I solved my issue: I was compiling with tigl3 but I was using tigl2 at runtime. I proved to myself that tigl3 java binding actually works by creating a separate project for testing your `tigl/examples/java_demo/JavaDemo.java` and it works as expected.
And it works correctly in my bigger Java project as well.

I think you can close this issue.

closed 2020-03-03 by rainman110