DLR-SC/tigl #659: Enable kinks in (fuselage) profiles

rainman110, 2020-03-06:
For many configurations, modeling kinks will be an important feature. 

The feature should allow:
 - Define kinks at a specified interpolation point
 - Define parameters, at which the points are interpolated. This allows, that e.g. kinks of two profiles meet exactly and adds more freedom to the profile interpolation.
 - Both require a CPACS defintion change of the profiles

rainman110, 2020-03-06:
Here's a first snapshot, of how the current status is:
[image]

@rmaierl, @sdeinert Is this, what you are looking for?

sdeinert, 2020-03-09:
Looks good!

rainman110, 2020-03-10:
@sdeinert, @rmaierl, @MarAlder I have a first (hopefully) working version at the branch https://github.com/DLR-SC/tigl/tree/profiles_with_kinks

There is also an example file https://github.com/DLR-SC/tigl/blob/profiles_with_kinks/tests/unittests/TestData/simpletest-fuselagekinks.cpacs.xml, which demonstrates the concept.

closed 2020-05-12 by rainman110