DLR-SC/tigl #662: Issue with get_volume() function of TiGL configuration manager after fuselage modification

VivTrif, 2020-03-12:
Hello,

In a loop that requires to modify the length or max width of the fuselage and trying to use the get_volume() function, an error occurs and the program crashes.

This seems to appear in a pseudo-random way, as by running the script on two different computers we had the crash ocuring at the same iteration numbers.

The modifications are made with the cpacscreator version of Tigl.

Error message from running it in the console :
  return _configuration.CCPACSFuselage_get_volume(self)
  RuntimeError: OpenCASCADE Error: TopoDS_Builder::Add

Error message when running a script :
  tigl3.tigl3wrapper.Tigl3Exception: TIGL_ERROR (1) [('tiglFuselageGetVolume', 1), {}] {}
  capi_return is NULL

I joined a script that triggers the error and a CPACS file to run the script with. With these files the error appears at iterations [11, 13, 29, 33, 46, 49, 65, 66, 80, 95].

Thanks in advance for taking a look at it !

[D150_AGILE_Hangar_v3.1.xml.zip](https://github.com/DLR-SC/tigl/files/4325104/D150_AGILE_Hangar_v3.1.xml.zip)

aidanjungo, 2020-03-18:
I just ran the script on macOS Catalina, I got only 3 time the error at iterations [25,64,91]

aidanjungo, 2020-07-01:
@rainman110 @joergbrech Do you also get this error? Apparently it is not specific to this function, it also happened with other geometric functions (like CCPACSFuselage_get_length). It think it may come from QT, but I don't know how to test that.

joergbrech, 2020-07-13:
@aidanjungo and @VivTrif, I am really sorry we kept you waiting so long. I will check out your script this week and see if I can reproduce the error.

joergbrech, 2020-07-13:
Ok, so I can reproduce the bug, allbeit with yet a different set of iteration numbers. Sorry again for disregarding this issue for so long and thanks for reporting it here!

The issue is related to #582 and fixed with https://github.com/DLR-SC/tigl/commit/118e64079723d306ef5f3eca65a4cace1552655a, which is not in the `cpacs_creator_dev` branch (yet).

So we should 
 - either cherrypick the workaround into `cpacs_creator_dev` or merge `master` into `cpacs_creator_dev` to get it fixed
 - possibly create a patched cpacscreator release to Anaconda

rainman110, 2020-07-14:
Sorry, also from my side. I cherry-picked the bugfix 118e640 onto cpacs_creator_dev, so the bug should be fixed there.

Next steps would be, to make a new cpacs_creator_release based on TiGL 3.1

VivTrif, 2020-07-15:
Thanks a lot for fixing it, and no problem for the delay !

svengoldberg, 2024-12-04:
The CPACS Creator is now based on the most current TiGL version (3.4):
https://github.com/DLR-SC/tigl/tree/cpacscreator-v0.2.0

This should be fixed now

closed 2024-12-04 by svengoldberg