DLR-SC/tigl #664: GenericSystems not based on generated classes

RlanderRISCSW, 2020-03-24:
The classes `CCPACSACSystems`, `CCPACSGenericSystems` and `CCPACSGenericSystem` don't use the base classes generated by the code generator yet. Therefore functionality for adding or removing systems during runtime is missing.
I implemented a proposal for changing to the generated base classes on the branch `generated_systems`, but get errors with the TravicCI/AppVeyor builds (which I can not reproduce on my machine). Note that the proposal would change the public API in the `CCPACSConfiguration` class since the `CCPACSACSystems` getter functions return an optional now.

rainman110, 2020-03-30:
I'll have a look why there are errors. Maybe you are using a different version of the generator?

rainman110, 2020-03-30:
Don't worry about the failing build. These are just the python bindings. We  have to adjust the swig interface files, such that it can deal with the new structure.

closed 2020-06-10 by joergbrech