DLR-SC/tigl #666: CTiglTopoAlgorithms::IsDegenerated fails on OCCT 7.2.0

rainman110, 2020-03-30:
RISC reported, that the unit test for this new function fails.

rainman110, 2020-03-30:
@RlanderRISCSW Which unit test is failing now?

RlanderRISCSW, 2020-03-31:
There was no unit test failing, but we got a problem with one of our test models from Descartes. The check for IsDegenerated incorrectly returned false for the front section of the fuselage. This occured when using OCCT 7.2.0, when building TIGL with OCE 0.17.2 the function works as expected.
The different behavior comes from `BRepBuilderAPI_MakeFace` which seems to be successful in OCCT 7.2.0, even though the front section is a vertex.
In Descartes we added a check for the edge length of the section wire, in case the length is (nearly) zero we take it as degenerated:

```
    double length = 0;
    TopoDS_Iterator iter(wire);
    for (; iter.More(); iter.Next())
    {
        const TopoDS_Edge& anEdge = TopoDS::Edge(iter.Value());
        if (!BRep_Tool::Degenerated(anEdge)) {
            isDegen = Standard_False;
        }
        length += GetLength(anEdge);
    }

    if (!isDegen && length < Precision::Confusion()) {
        isDegen = true;
    }
```
[cpacs_out.zip](https://github.com/DLR-SC/tigl/files/4409878/cpacs_out.zip)

rainman110, 2020-03-31:
Good idea!

RlanderRISCSW, 2020-03-31:
Sorry, the problem still resides since the final check for degenerated==true is still based on the output of `BRepBuilderAPI_MakeFace`, which behaves different in OCCT 7.2.0.

rainman110, 2020-03-31:
So you completely removed the BRepBuilderAPI_MakeFace check here?

RlanderRISCSW, 2020-04-01:
No, I did the length check before the BRepBuilderAPI_MakeFace check.

rainman110, 2020-04-01:
Could you please paste the code for the function?

RlanderRISCSW, 2020-04-01:
```
bool CTiglTopoAlgorithms::IsDegenerated(const TopoDS_Wire& wire)
{
    bool isDegen = Standard_True;

    double length = 0;
    TopoDS_Iterator iter(wire);
    for (; iter.More(); iter.Next())
    {
        const TopoDS_Edge& anEdge = TopoDS::Edge(iter.Value());
        if (!BRep_Tool::Degenerated(anEdge)) {
            isDegen = Standard_False;
        }
        length += GetLength(anEdge);
    }

    if (!isDegen && length < Precision::Confusion()) {
        isDegen = true;
    }

    if (!isDegen) {
        BRepBuilderAPI_MakeFace facemaker(wire);
        if (!facemaker.IsDone()) {
            isDegen = true;
        }
    }

    return isDegen;
}
```

rainman110, 2020-04-01:
Merci! In principle the ```!isDegen``` in front of the length check can be removed, right?

RlanderRISCSW, 2020-04-01:
yes, you are right, this is a bit redundant  :-)

rainman110, 2020-04-01:
f0be905 implements your code.

RlanderRISCSW, 2020-04-01:
perfect, now it works

commit 7c9ef422 (2020-03-31): Fixed bug in CTiglTopoAlgorithms::IsDegenerated

Also added a test for this issue

Fixes #666
commit 84888639 (2020-04-22): Fixed bug in CTiglTopoAlgorithms::IsDegenerated

Also added a test for this issue

Fixes #666
closed 2020-04-01 by rainman110