DLR-SC/tigl #685: tiglWingGetSegmentIndex raises exception under the hood

rainman110, 2020-05-25:
The function provokes an exception, if the segment is not on the first wing. This exception is caught by tigl and the next wing will be checked for the segment.

Even though, this is not necessary an error, using exception to control the program flow is an anti-pattern. Also, users of Visual Studio will see the exception, even thoug TiGL internally handles it.

joergbrech, 2020-05-26:
The same *catch-continue* pattern is used in 

 - [ ] `tiglWingComponentSegmentGetSegmentIntersection`
 - [ ] `tiglWingComponentSegmentComputeEtaIntersection` 
 - [ ] `tiglWingComponentSegmentGetNumberOfSegments`
 - [ ] `tiglWingComponentSegmentGetSegmentUID`
 - [ ] `tiglFuselageGetSegmentIndex`
 - [ ] `tiglWingComponentSegmentGetMaterialCount`
 - [ ] `tiglWingComponentSegmentGetMaterialUID`
 - [ ] `tiglWingComponentSegmentGetMaterialThickness`

😜

rainman110, 2020-05-27:
phew... a lot of brainless work :D

commit 21138a24 (2020-07-17): refactor: use uid manager instead of exceptions to find components

fixes #685
commit c9a5ece0 (2020-07-17): refactor: use uid manager instead of exceptions to find components

fixes #685
commit ca7acaf3 (2020-07-17): Fixed exception in tiglWingGetSegmentIndex

This changes the code in tiglWingGetSegmentIndex
to not internally use exceptions for the control flow.

Fixes #685
PR #699 (2020-07-18): Refactor: Don't use exceptions to obtain references to geometric components

## Description

Fixes #685. Check the issue description for more details.

## Checklist:

- [ ] A test for the new functionality was added.
- [x] All tests run without failure.
- [x] The new code complies with the TiGL style guide.
- [ ] New classes have been added to the Python interface.
- [ ] API changes were documented properly in tigl.h.
closed 2020-07-18 by rainman110