DLR-SC/tigl #689: Confusion about default behavior of `tiglWingGetUpperPoint` and `tiglWingGetLowerPoint`

joergbrech, 2020-06-15:
Currently the default behavior is to interpret the arguments as parameters on the actual loft, rather than eta-xsi coordinates along the trapezoidal chord face. 

Unfortunately, the TiGL documentation is contradictory:
 - The [html documentation](https://dlr-sc.github.io/tigl/doc/latest/group__WingFunctions.html#ga0522075fa74baa1c844c841a545397c5) on the TiGL Website correctly documents this.
 - The html documentation distributed in the [Github Release asset](https://github.com/DLR-SC/tigl/releases/download/v3.0.0/TIGL-3.0.0-win64.zip) does not mention the default behavior. **Why is there a different documentation on the website and in the release?**
 - The [Changelog](https://dlr-sc.github.io/tigl/doc/latest/md_build-tigl-Desktop_Qt_5_10_1_MSVC2015_64bit-Default_doc_ChangeLog.html) falsely claims that the eta-xsi coordinates along the trapezoidal chord face are the default interpretation

It can be argued which behavior would be better suited as default. Using `onLinearLoft` would be backwards compatible. `asParameterOnSurface` respects the actual loft of the wing, so it would be more intuitive.

rainman110, 2020-06-15:
Really? I assumed that ``onLinearLoft`` is default because of backwards compatibility.

joergbrech, 2020-06-15:
Nope:

https://github.com/DLR-SC/tigl/blob/0653b2342efa86495f852190deda423ad71bfd3f/src/wing/CCPACSWingSegment.h#L288

I guess we should decide once and for all and make sure that the documentation is clear about this.

joergbrech, 2020-07-16:
> It can be argued which behavior would be better suited as default. Using onLinearLoft would be backwards compatible. asParameterOnSurface respects the actual loft of the wing, so it would be more intuitive.

We now agreed to keep the default behavior as it currently is: **The inputs are interpreted as parameters on surface by default**. There are good arguments for both choices. In the end, using parameters on the surface is less likely to cause errors for wings with guide curves: With the other option, not every point on a wing might be reachable using an orthogonal projection from the chordface.

Thus, only the documentation must be cleaned:
 - [ ] Fix Changelog on the Releases page
 - [ ] Fix Changelog in the html documenation for TiGL 3.0

closed 2020-07-17 by joergbrech