DLR-SC/tigl #693: Problems with build in Ubuntu 20.04

e-dub, 2020-06-28:
I built tixi and tigl in Ubuntu 20.04 and although it appears to have successfully compiled, I am having trouble accessing tigl from Python. 

I built the software with the following code (which includes adding `tigl3/tigl3wrapper.py` and `tixi3/tixi3wrapper.py` to site-packages). 

```
cd ~/opt
git clone https://github.com/DLR-SC/tixi
cd tixi
mkdir build
cd build
cmake ..                                    \
-DBUILD_SHARED_LIBS=ON                      \
-DPYTHON_EXECUTABLE=/usr/bin/python3.8
make -j4
sudo make install
sudo mkdir /usr/lib/python3.8/dist-packages/tixi3
sudo cp /usr/local/share/tixi3/python/* /usr/lib/python3.8/dist-packages/tixi3

cd ~/opt
git clone https://github.com/DLR-SC/tigl
cd tigl
mkdir build
cd build
cmake ..                                    \
-DBUILD_SHARED_LIBS=ON                      \
-DPYTHON_EXECUTABLE=/usr/bin/python3.8     \
-DCMAKE_INSTALL_PREFIX=install
make -j4
sudo make install
sudo cp -r /usr/local/share/tigl3/python/tigl3/ /usr/lib/python3.8/dist-packages/
```
When I try to run the Python examples I get the following errors:
```
Traceback (most recent call last):
  File "example_fuselage_structure.py", line 5, in <module>
    import tigl3.configuration
ModuleNotFoundError: No module named 'tigl3.configuration'
```
and
```
Traceback (most recent call last):
  File "example_visualization.py", line 5, in <module>
    import tigl3.configuration, tigl3.geometry, tigl3.boolean_ops, tigl3.exports
ModuleNotFoundError: No module named 'tigl3.configuration'
```

Any ideas what I did wrong?  I am not seeing it.

Thanks in advance!

rainman110, 2020-06-29:
Hi @e-dub . TiGL comes with two flavors of python extensions
 - C API Bindings
 - C++ API Bindings

The C Bindings are created with the standard build and results in a tigl3wrapper module.

What you want to use are the C++ Bindings, which have to be activated manually during the build process. To make your live easier, just install tigl from conda. If you still want to build it yourself, continue reading ;):

First, before trying to build it, make sure that you meet the dependencies - also with their exact version:
 - SWIG
 - PythonOCC 0.17.x + sources
 - OCE 0.17.x compiled as dynamic shared libraries (*.so files)
 - tixi3

As you might see, a newer OpenCASCADE Version is not yet supported. When configuring cmake, add also the following arguments:

```
-DPythonOCC_SOURCE_DIR=/src_to/pythonocc-core_0.17.x
-DTIGL_BINDINGS_PYTHON_INTERNAL=ON
```

TiGL build process requires to find the source of pythonocc, to allow interoperability between tigl and pythonocc.

After building and installing TiGL, you finally need to copy manually the installed tigl3 package from ``install_prefix/share/tigl3/python`` to your python-site-packages folder.

Look at our conda recipe, how we do this automatically:
https://github.com/DLR-SC/tigl-conda/blob/master/tigl3/build.sh

e-dub, 2020-07-01:
Besten Dank!

I have now rebuilt pythonOCC with
```
cd ~/opt 
git clone git://github.com/tpaviot/pythonocc-core.git
cd pythonocc-core
git checkout 0.18.1
mkdir build
cd build
cmake ..                                                            \
-DBUILD_SHARED_LIBS=ON                                              \
-DPYTHONOCC_WRAP_SMESH=ON                                           \
-DPYTHONOCC_INSTALL_DIRECTORY=/usr/lib/python3.8/dist-packages/OCC  \
-DPYTHON_EXECUTABLE=/usr/bin/python3.8                              \
-DPYTHON_INCLUDE_DIR=/usr/include/python3.8                         \
-DPYTHON_LIBRARY=/usr/lib/x86_64-linux-gnu/libpython3.8.so 
make -j4
sudo make install
```
and then tigl3 with
```
cd ~/opt
git clone https://github.com/DLR-SC/tigl
cd tigl
mkdir build
cd build
cmake ..                                                \
-DBUILD_SHARED_LIBS=ON                                  \
-DPYTHON_EXECUTABLE=/usr/bin/python3.8                  \
-DCMAKE_INSTALL_PREFIX=install                          \
-DPythonOCC_SOURCE_DIR=$HOME/opt/pythonocc-core/     \
-DTIGL_BINDINGS_PYTHON_INTERNAL=ON
make
sudo make install
sudo cp -r /usr/local/share/tigl3/python/tigl3/ /usr/lib/python3.8/dist-packages/
```
The compiling is definitely doing more and again finishes without errors.  Unfortunately, I still have the same errors as above when trying to call the examples.  Any further ideas for me?

Grüße aus Bozen

rainman110, 2020-07-01:
I can jst guess now. To help me, please upload the complete build log here (you can drag in text files etc). Also add the error message that you get from python.

e-dub, 2020-07-01:
Python error:```

Traceback (most recent call last):
  File "example_fuselage_structure.py", line 5, in <module>
    import tigl3.configuration
ModuleNotFoundError: No module named 'tigl3.configuration'
```

[cmakeout.txt](https://github.com/DLR-SC/tigl/files/4858295/cmakeout.txt)
[makeout.txt](https://github.com/DLR-SC/tigl/files/4858296/makeout.txt)
[installout.txt](https://github.com/DLR-SC/tigl/files/4858298/installout.txt)
[CMakeCache.txt](https://github.com/DLR-SC/tigl/files/4858302/CMakeCache.txt)
[install_manifest_.txt](https://github.com/DLR-SC/tigl/files/4858309/install_manifest_.txt)

rainman110, 2020-07-01:
What about the build log? The files are not really attached

e-dub, 2020-07-01:
It made a link for some reason:

[](https://github.com/DLR-SC/tigl/files/4858296/makeout.txt)

rainman110, 2020-07-01:
Hm... this all look pretty well, as it should be. Can you also attach the directory content of ``/usr/lib/python3.8/dist-packages/tigl3`` and ``/usr/lib/python3.8/dist-packages/tixi3`` ?

What output do you get for the following command?
```sh
ldd /usr/lib/python3.8/dist-packages/tigl3/_configuration.so
```

e-dub, 2020-07-01:
I guess that is indeed the problem 
/usr/local/share/tigl3/python/tigl3 only has tigl3wrapper.py and therefore I am only copying that file to /usr/lib/python3.8/dist-packages/tigl3

The files I beleive I want to have are in the folder ~/opt/tigl/build/bindings/python_internal

```
boolean_ops.py
_boolean_ops.so
CMakeFiles
cmake_install.cmake
configuration.py
_configuration.so
core.py
_core.so
doc
doc.i
Doxyfile
exports.py
_exports.so
geometry.py
_geometry.so
imports.py
_imports.so
Makefile
occtaddons.py
_occtaddons.so
tmath.py
_tmath.so

```
Is that correct?

Thanks for your time!

e-dub, 2020-07-01:
Ok I have solved it! 
I removed the following line:
-DCMAKE_INSTALL_PREFIX=install   
This installed everything in the build folder. Without this, the Python files were saved to /usr/lib/python3.8/dist-packages/tigl3 as expected.  I am now rocking and rolling with tigl3 examples. 

Thank you for your help!

rainman110, 2020-07-01:
Great that you solved it. Another solution would have to to copy the modules from the correct install path ;) :
```
sudo cp -r /home/w/opt/tigl/build/install/share/tigl3/python/tigl3/ /usr/lib/python3.8/dist-packages/ 
```

You were using the wrong source path.

rainman110, 2020-07-01:
please reopen the issue, if the problems reappears.

closed 2020-07-01 by rainman110