DLR-SC/tigl #700: TiGL Crash in CTiglPatchShell

joergbrech, 2020-07-20:
In seldom cases, TiGL crashes if the following assertion fails, e.g. if the ouptut of `BRepBuilderAPI_Sewing` is of type `TopAbs_COMPOUND`:

https://github.com/DLR-SC/tigl/blob/a1fc696bc623ed270e98bec456631475aa32ec80/src/geometry/CTiglPatchShell.cpp#L169

I don't know why/when this happens, but until find a fix to make our code more robust, we should at least make sure that TiGL doesn't crash hard.

joergbrech, 2020-07-20:
Here is one example where this happens:  [Bo105_v0_04_maserati_cpacs3.zip](https://github.com/DLR-SC/tigl/files/4946667/Bo105_v0_04_maserati_cpacs3.zip) *(I found it browsing some of the old TiGL issues: #198)*.

rainman110, 2020-07-20:
Strange, I was certain that at least the example works fine with TiGL. I agree, that TiGL should not crash.

Most of these errors seemed to happen, when the closing faces (fron / back) could not be built

joergbrech, 2020-07-20:
To be fair, it is a CPACS 2 rotorcraft and the issue refers to TiGL 2.1. I just auto-updated the file and checked if the issue might magically be fixed in TiGL 3. So there could still be quite a few issues with the cpacs file, I haven't taken a detailed look yet. 

In any case, we should encapsulate the code in a try-catch block for now.

rainman110, 2020-07-20:
already have a fixed version

rainman110, 2020-07-20:
[Bo105_v0_04_maserati.fixed.zip](https://github.com/DLR-SC/tigl/files/4946745/Bo105_v0_04_maserati.fixed.zip)

rainman110, 2020-07-20:
When do you experience the crash? TiGL Viewer shows an error for uid="Fuselage_struct", but it doesn't crash

rainman110, 2020-07-20:
[Bo105_v0_04_maserati-v3.zip](https://github.com/DLR-SC/tigl/files/4946768/Bo105_v0_04_maserati-v3.zip)

(I am wondering, whether this is the same file than the file you posted)

rainman110, 2020-07-20:
@joergbrech Strange, my file does not crash, your's does. So the diff between both could give a hint about the crash reason.

rainman110, 2020-07-20:
@joergbrech There are strange scaling values in your file. The all are string, probably to a wrong cpacs conversion.

Still, crashing should be avoided.

joergbrech, 2020-07-20:
> Update: Looks also like a bug in cpacs2to3

My CPACS 2 file had a `uID="1" in it, I think `cpacs2to3` autoreplaced all occurences`1` with the automatically generated `uID`.

joergbrech, 2020-07-20:
See DLR-SC/cpacs2to3#14

PR #701 (2020-07-21): avoid hard crash in CTiglPatchShell
Minor fix to prevent a hard crash in `CTiglPatchShell`. Fixes #700
closed 2020-07-21 by rainman110