DLR-SC/tigl #713: tigl3 installation with conda raise an error importing tigl3wrapper in python

condoratberlin, 2020-07-27:
while importing tigl3wrapper an error message raised:
```
Exception: Could not load the tigl3 library. Please check if:
1) The PATH (Windows) / LD_LIBRARY_PATH (Linux) environment variable points to the library
2) The architecture of the library matches the architecture of python (a 32 bit python needs a 32 bit shared library)
```
The Library can be found manually in `.../anaconda3/envs/env_name/lib`

The Problem seems to be linux specific.

rainman110, 2020-07-27:
Could you please add some context?

 - How did you install tigl3 (which conda command?)
 - How did you try to import the library?
 - What python version are you using?
 - What os / linux is affected?

Actually, importing is actually tested by our conda build scripts.  Thereore some more information would be really helpful.

joergbrech, 2020-07-27:
We just released a new tixi version (the xml-parser used by tigl) which does not seem to be binary compatible to the latest tigl release and I am guessing that conda installed the new tixi 3.1.1 release. You can verify this by typing `conda list`.

A workaround would be to downgrade tixi:

```
conda install tixi3=3.0.3 -c dlr-sc
```

With DLR-SC/tigl-conda@8c454ed this should be fixed in the future. Thanks for letting us know!

joergbrech, 2020-07-27:
Could you please also let us know if this resolves your issue? Thanks

condoratberlin, 2020-07-27:
Sure:
I figured out, that tixi3 seems to be the reason.
 I use an environment.yml file:
```
name: test
channels:
  - conda-forge
  - dlr-sc
  - anaconda
  - defaults
dependencies:
  - nlopt
  - python=3.7
  - pip
  - numpy
  - matplotlib
  - pandas>=0.25
  - h5py
  - dask
  - distributed
  - numba
  - pytest
  - pytest-cov
  - scipy
  - openpyxl
  - tixi3
  - tigl3
  - h5py
  - sphinx
  - sphinx_rtd_theme
  - mathjax
  - nbsphinx
  - jupyter_client
  - ipykernel
  - pip:
      - anytree==2.7.2
      - pytest-datadir
```
Install with:
```
conda env create -f environment.yml
conda activate test

```
if i fix tixi3 to be version 3.0.3 (tixi3=3.0.3)everything works as before. Starting from Version tixi3=3.1.0 it fails.
```
$ python
Python 3.7.8 | packaged by conda-forge | (default, Jul 23 2020, 03:54:19) 
[GCC 7.5.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from tigl3 import tigl3wrapper
Traceback (most recent call last):
  File "/home/daeh_sa/anaconda3/envs/lightworks/lib/python3.7/site-packages/tigl3/tigl3wrapper.py", line 294, in Tigl3
    lib = ctypes.CDLL(os.path.join(sys.prefix, 'lib', 'libtigl3.so'))
  File "/home/daeh_sa/anaconda3/envs/lightworks/lib/python3.7/ctypes/__init__.py", line 364, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: libtixi3.so.0: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/daeh_sa/anaconda3/envs/lightworks/lib/python3.7/site-packages/tigl3/tigl3wrapper.py", line 281, in <module>
    class Tigl3(object):
  File "/home/daeh_sa/anaconda3/envs/lightworks/lib/python3.7/site-packages/tigl3/tigl3wrapper.py", line 297, in Tigl3
    "  1) The PATH (Windows) / LD_LIBRARY_PATH (Linux) environment variable points to the library\n" +
Exception: Could not load the tigl3 library. Please check if:
  1) The PATH (Windows) / LD_LIBRARY_PATH (Linux) environment variable points to the library
  2) The architecture of the library matches the architecture of python (a 32 bit python needs a 32 bit shared library)

>>> 

```
It fails on fedora31 and docker miniconda container used for continous integration.

condoratberlin, 2020-07-27:
Yes it resolves the issue!

joergbrech, 2020-07-27:
I will close this, as the conda package for TiGL 3.0 is pinned to tixi3==3.0.3 now. Thanks for the find! Note that we will soon (in the next hours or days) publish TiGL 3.1 based on the new tixi release.

closed 2020-07-27 by joergbrech