DLR-SC/tigl #718: CI cache issues

rainman110, 2020-07-27:
CI assumes matlab and doxygen, if the cache is available.

Sometimes, the download however failed and therefore they are also not in the cache.

Hence, we should better check directly, which tool is there and which is not and then download them if required.

Otherwise, it is impossible to download the tools by retrigger the build

joergbrech, 2020-07-27:
That's weird, I assumed that the cache is only stored if the job is successfull. What do you propose? Should we disable the caching of the windows dependencies?

If caching is so errorporne, I think that would be reasonable, since we are already trying to download them up to 9 times with a 10 minute time out (lines 142, 145, 149 and 151):

https://github.com/DLR-SC/tigl/blob/cb91d44bf9de6db3e06d6cf4fd1750fbe10558d7/.github/workflows/build-test-deploy.yml#L140-L152

joergbrech, 2020-07-27:
There is still an issue: The doxygen executable on windows is *never* found and therefore the target doc does not exist, which should be built at package creation. Sometimes this causes the step to fail, but most of the times it doesn't and the CI logs look fine with calming green checkmarks...

rainman110, 2020-07-27:
Very strange behavior.

Then, I’ll try to create custom doxygen conda packages that will hopefully work.

joergbrech, 2020-07-28:
My guess is that there is a problem with the CMAKE_PREFIX_PATH, rather than with the package

PR #719 (2020-07-28): Remove thirdparty CI cache
The CI Cache of doxygen and matlab for windows does not work properly: Sometimes an empty directory is cached after the download failed.

Fixes #718
commit 7ad22c7d (2020-07-28): Remove thirdparty CI cache

The CI Cache of doxygen and matlab for windows does not work properly: Sometimes an empty directory is cached after the download failed.

Fixes #718
closed 2020-07-28 by rainman110