DLR-SC/tigl #730: TiGL wiki may has misleading information.

Mk-arc, 2020-08-05:
In the wiki section "Building TiGL from source" it is recommended to download the dependencies from the packet repo via 

```
echo 'deb http://download.opensuse.org/repositories/science:/dlr/xUbuntu_16.04/ /' >> /etc/apt/sources.list.d/tigl.list 
sudo apt-get update
```
However.. the server seems to not have a valid Certificate, which makes this way of installing the dependencies not feasible.

rainman110, 2020-08-05:
You are right, it has to be:

```bash
echo 'deb http://download.opensuse.org/repositories/science:/dlr/xUbuntu_16.04/ /' | sudo tee /etc/apt/sources.list.d/science:dlr.list
curl -fsSL https://download.opensuse.org/repositories/science:dlr/xUbuntu_16.04/Release.key | gpg --dearmor | sudo tee /etc/apt/trusted.gpg.d/science:dlr.gpg > /dev/null
sudo apt update
```

Mk-arc, 2020-08-05:
> You are right, it has to be:
> 
> ```shell
> echo 'deb http://download.opensuse.org/repositories/science:/dlr/xUbuntu_16.04/ /' | sudo tee /etc/apt/sources.list.d/science:dlr.list
> curl -fsSL https://download.opensuse.org/repositories/science:dlr/xUbuntu_16.04/Release.key | gpg --dearmor | sudo tee /etc/apt/trusted.gpg.d/science:dlr.gpg > /dev/null
> sudo apt update
> ```

I added this to the wiki page. I didnt except that I can simply edit it without any review as an extern :D

rainman110, 2020-08-05:
Oh... did not know this!

closed 2020-08-11 by Mk-arc