DLR-SC/tigl #739: Wrong control surface count results

AlejandroRios, 2020-09-03:
Hi!

Tigl control surfaces count returns discrepant results. I'm using the D150_v30.xml as reference file. When applying:

tigl.getControlSurfaceCount('D150_wing_CS')

got 3 as a result, but this model has trailing edge devices (3) and spoilers (5). Therefore, I was expecting 8 control surfaces detected. It seems like tigl only performs the count of trailing edge devices.
When applying:

tixi.getNumberOfChilds(model_xpath+ 'wings/wing[1]/componentSegments/componentSegment/controlSurfaces/trailingEdgeDevices/')

and 

tixi.getNumberOfChilds(model_xpath+ 'wings/wing[1]/componentSegments/componentSegment/controlSurfaces/spoilers/')

I got the expected results. An I doing something wrong?

rainman110, 2020-09-03:
Hi! Currently only trailing edge devides are supported, therefore only those are counted.

We plan to add leading edge devices also to TiGL. Spoiler are not yet planned as we don't have a good idea, how to model these.

AlejandroRios, 2020-09-03:
Thanks for the quick reply!

In that case, I think that for now, the tixi solution works for my application.

Best regards!

closed 2020-10-19 by rainman110