DLR-SC/tigl #741: Errors in wingComponentSegmentComputeEtaIntersection

sfreund-DLR, 2020-09-22:
The function wingComponentSegmentComputeEtaIntersection returns wrong xsi-coordinates when eta=0.

tigl version 3.1.0

Test was done with the python file [tigl_test.txt](https://github.com/DLR-SC/tigl/files/5260517/tigl_test.txt) and using the configration from [tigl-examples](https://raw.githubusercontent.com/DLR-SC/tigl-examples/master/cpacs/CPACS_30_D150.xml).

rainman110, 2021-04-23:
I agree, this is a bug. I'll look into it now.

rainman110, 2021-04-23:
@joergbrech I cannot reproduce these errors from @sfreund-DLR for the case eta=0. 

I have some error of max. 2e-3 on the iso line, as reported though.

rainman110, 2021-04-23:
So here's my first analysis / conclusion.

The function `tiglWingComponentSegmentComputeEtaIntersection` is meant to draw a straight line between the two specified points on the component segment. This line is then intersection with an iso-eta line to get the xsi coordinate.

This works well / accurate, as long as the cord surface is planar. In the D150 example, it is twisted. __This means, that the straight line is not identical to a true iso-xsi line!!!__ Since both lines differ, it is clear that the result is not exact.

@sfreund-DLR I know that this answer is probably not what you wanted to hear. The function is designed like this by intention. There are examples, where the line passes segment borders. The function's primarily use is to create straight edges for leading and trailing edge devices.

rainman110, 2021-04-23:
I am still puzzled, why you get these 0.5 numbers.

rainman110, 2021-04-23:
Update: I can reproduce the original strange numbers with TiGL 3.0.0.

Using TIGL from master, i get the following outout:

```
tigl version 3.1.0
expected: [0.  0.1 0.2 0.3 0.4 0.5 0.6 0.7 0.8 0.9 1. ]
got:  [2.9802322107863443e-08, 0.09999999999999985, 0.19999999999999982, 0.30000000000000004, 0.3999999999999999, 0.5, 0.6000000000000003, 0.6999999999999998, 0.8, 0.8999999999999999, 0.9999999999999997]
[0.3999999999999999, 0.4000899236633792, 0.4001798473267589, 0.400269770990138, 0.4003596946535174, 0.40044961831689646, 0.4005395419802763, 0.4006294656436553, 0.4007193893070347, 0.400809312970414, 0.39999999999999997]

```tigl version 3.1.0
expected: [0.  0.1 0.2 0.3 0.4 0.5 0.6 0.7 0.8 0.9 1. ]
got:  [2.9802322107863443e-08, 0.09999999999999985, 0.19999999999999982, 0.30000000000000004, 0.3999999999999999, 0.5, 0.6000000000000003, 0.6999999999999998, 0.8, 0.8999999999999999, 0.9999999999999997]
[0.3999999999999999, 0.4000899236633792, 0.4001798473267589, 0.400269770990138, 0.4003596946535174, 0.40044961831689646, 0.4005395419802763, 0.4006294656436553, 0.4007193893070347, 0.400809312970414, 0.39999999999999997]

rainman110, 2021-04-23:
As I said, the telorance issue cannot be fixed due to the way the function is intended to work. The 0.5 values are fixed now.

Therefore, I suggest du close this issue if @sfreund-DLR agrees.

rainman110, 2021-04-23:
@joergbrech Could you please have a look to my comments? How should we proceed?

joergbrech, 2021-04-26:
> This means, that the straight line is not identical to a true iso-xsi line!!! Since both lines differ, it is clear that the result is not exact.

Sounds reasonable.

> @joergbrech Could you please have a look to my comments? How should we proceed?

@rainman110 , @sfreund-DLR, I suggest keeping this issue open a little longer in case there is still need for a discussion, but I would remove the TiGL 3.2 label. I don't think anything needs to be done for the next TiGL release.

rainman110, 2021-04-26:
I again thought about this. The iso-xsi line should be a straight line. I'll look again into it.

rainman110, 2021-04-26:
Now I have it. The problem is, that eta=0,10398 is already somewhat beyond the segment border. Hence, the iso-xsi line has a kink and is not a straight line:

[image]

Zooming in:
[image]

Therefore, you can see these small "errors". If you closely look, you can see that the differences to you expected values increase linearly, which now makes sense to me.

For me, this is not a bug but the function works like expected. To fix your tolerences, you have to use a smaller (better) eta value, that is exactly on the segment border.

sfreund-DLR, 2021-04-28:
thanks @rainman110 and @joergbrech for investigating the issues. 

starting with my **second issue**. Sorry somehow I had a "bad" eta coordinate. I obtained it using several TIGL functions since I did not find a direct way to get from a segment border to componentSegment eta/xsi. So do you think, there should be a function wingID, SegmentID, SegmentEta, SegmentXsi --> CompSegEta, CompSegXsi? Now I used tiglWingGetChordPoint and tiglWingComponentSegmentPointGetEtaXsi.

**first issue**
in my [miniforge environment](https://github.com/DLR-SC/tigl/files/6392929/environment_py37_tigl3.txt), the issue still exists. I tried it with 
- tigl version 3.1.0 tixi version 3.1.1 (see env file)
- tigl version 3.0.0 tixi version 3.1.0

Using this slightly modified python file, [tigl_test.txt](https://github.com/DLR-SC/tigl/files/6392936/tigl_test.txt), it seems that when eta2 is located in the twisted segment (eta2=0,10398), still 0.5 values arise. I have the same behavior for both tixi/tigl versions tested. I also tried the tigl zip file instead of the anaconda-package with the same result.

rainman110, 2021-04-28:
We are releasing Tigl 3.2 in the next couple of days. The first issue will be fixed there.

Although my script reported TiGL 3.1.0, it was a more recent development version.

sfreund-DLR, 2021-04-29:
cool, I didn't get this one. Then the issue can be closed. Thanks a lot!

closed 2021-09-28 by joergbrech