DLR-SC/tigl #742: weird linking issue

kurtsansom, 2020-09-24:
I am able to build but the library linking is not working in a weird way.

I am having a weird build/linking issue that appears to link some of the opencascade libraries but not all of them. somehow it cannot find all of them. If it didn't find any of them I would understand, but that it can't find three of them seems super strange.

If I export LD_LIBRARY_PATH to the opencascade library paths and then it subsequently works, but I really get annoyed with that as its kinda a cheat.

truncated output:
```
ldd ./examples/c_demo/c_demo 
	libTKIGES.so.7 => /home/sansomk/build/opencascade/install/lib/libTKIGES.so.7 (0x00007f2ded9dc000)
	libTKSTL.so.7 => /home/sansomk/build/opencascade/install/lib/libTKSTL.so.7 (0x00007f2ded7c2000)
	libTKSTEP.so.7 => /home/sansomk/build/opencascade/install/lib/libTKSTEP.so.7 (0x00007f2ded27b000)
	libTKOffset.so.7 => /home/sansomk/build/opencascade/install/lib/libTKOffset.so.7 (0x00007f2deced9000)
	libTKMesh.so.7 => /home/sansomk/build/opencascade/install/lib/libTKMesh.so.7 (0x00007f2decbef000)
	libTKSTEPBase.so.7 => /home/sansomk/build/opencascade/install/lib/libTKSTEPBase.so.7 (0x00007f2dec75f000)
	libTKBool.so.7 => /home/sansomk/build/opencascade/install/lib/libTKBool.so.7 (0x00007f2dec12a000)
	libTKXSBase.so.7 => /home/sansomk/build/opencascade/install/lib/libTKXSBase.so.7 (0x00007f2debc5e000)
	libTKShHealing.so.7 => /home/sansomk/build/opencascade/install/lib/libTKShHealing.so.7 (0x00007f2deb7bd000)
	libTKBO.so.7 => /home/sansomk/build/opencascade/install/lib/libTKBO.so.7 (0x00007f2deb31c000)
	libTKPrim.so.7 => /home/sansomk/build/opencascade/install/lib/libTKPrim.so.7 (0x00007f2deb0b5000)
	libTKTopAlgo.so.7 => /home/sansomk/build/opencascade/install/lib/libTKTopAlgo.so.7 (0x00007f2deabc2000)
	libTKGeomAlgo.so.7 => /home/sansomk/build/opencascade/install/lib/libTKGeomAlgo.so.7 (0x00007f2dea4a0000)
	libTKBRep.so.7 => /home/sansomk/build/opencascade/install/lib/libTKBRep.so.7 (0x00007f2dea18e000)
	libTKGeomBase.so.7 => /home/sansomk/build/opencascade/install/lib/libTKGeomBase.so.7 (0x00007f2de9aa3000)
	libTKG3d.so.7 => /home/sansomk/build/opencascade/install/lib/libTKG3d.so.7 (0x00007f2de9777000)
	libTKG2d.so.7 => /home/sansomk/build/opencascade/install/lib/libTKG2d.so.7 (0x00007f2de9519000)
	libTKMath.so.7 => /home/sansomk/build/opencascade/install/lib/libTKMath.so.7 (0x00007f2de909d000)
	libTKernel.so.7 => /home/sansomk/build/opencascade/install/lib/libTKernel.so.7 (0x00007f2de8ccf000)
	libTKSTEPAttr.so.7 => not found
	libTKSTEP209.so.7 => not found
	libTKFillet.so.7 => not found
```

OpenCascade: 7.4.0.dev (snapshot from gitweb opencascade website)
cmake: 3.18.20200924 with CMAKE_INSTALL_RPATH_USE_LINK_PATH=TRUE
compiler: gcc-10.1.0
OS: Linux mint 19.3 (Ubuntu 18.04 base)

rainman110, 2020-09-29:
I can totally reproduce this weird issue. I don't yet fully understand, whats going on. It seems, that libTKSTEPAttr is an implicit dependency, meaning tigl does not directly depend on it but libTKSTEP. As it is not used, it is also not linked. During the startup, the demo still needs to load the transitive dependency.
Loading via RPATH does not work, as we would have to add the RPATH to all OpenCASCADE libs.

rainman110, 2020-10-17:
The reason is , that some opencascade libraries itself want to load e.g. its  own dependencies such as e.g. libTKSTEPAttr.so. 
Since opencascade's libs don't use an RPATH, these libraries cannot be resolved.

This issue must be fixed by opencascade. Therefore, I am closing this issue here.

closed 2020-10-17 by rainman110