DLR-SC/tigl #745: Issue with combination of guide curves and fuselage kinks

RlanderRISCSW, 2020-10-01:
When trying to add guide curves to a model containing fuselage kinks the kink edges are not generated any more. Additionally it seems that the curve parameters defined by the kinks are not taken into account by the guide curve. Is this a bug or is the combination of guide curves and kinks not supported?

Here the example I used (cpacs files in archive): [kinks_GC_issue.zip](https://github.com/DLR-SC/tigl/files/5312827/kinks_GC_issue.zip)

The left image shows the original fuselage with the kink defined in the front profile at parameter 0.3 (kink_noGC.xml). The right image shows the same model with a guide curve added at parameter 0.28.
[image][image]

It looks like the kink edge is not generated when a guide curve is defined, and also the changed parameter for the kink point in the first section (0.3) ist not taken into account by the guide curve (parameter 0.28 should be above the symmetry plane in the first section)

rainman110, 2020-10-05:
In theory, kinks and guidecurves should be compatible. I have to look into it, what causes the bug.

If you look closely, there still is the kink at param = 0.3. But the surface it not subdivided at the parameter. The reason is, that both surface creation mechanism are inherent different and this case was simply not taken into account.

rainman110, 2021-12-14:
Here's my analysis about the issue.

Actually, I see here two problems. One is a true bug, the other one is an inconvenience.

1) Bug: The guide curve is placed a parameter = 0.28, the kink at 0.3. Thus, the guide curve should be placed above the kink. In fact, it is currently placed below. It seems, that the positioning of the guide curve is executed before applying the kink parameters.

2) Inconvenience: The surface is not splitted at the kink. The reason is, the we currently have implemented two different face modes: First, split faces along guide curves. Second, split faces along kinks. A mix of both needs to  be implemented.

svengoldberg, 2024-06-12:
The first problem rather looks to me like an inconsistent usage within CPACS/TiGL... When calling the function `tigl.fuselageGetPoint(1,1, 0, 0.28)`, the guide curve's starting point is indeed hit. But it looks like, the kink is not set at parameter 0.3 (as you mentioned @rainman110). 
Hence, it rather seems like the the surface is reparameterized based on the arc length (which should not happen of course), since the CPACS definition of the guide curve's start/end is based on that (and is set to 0.28). Nevertheless, the curve's (-> profile's) parameterization seems not to be touched. 

After some investigations with @joergbrech, we found out the place at the source code where that happens (`tiglcommonfunctions::WireGetPointTangent(<...>)`). Here, the guide curve's starting point on the first profile is set to the value `0.28*arc_length`. That does not correspond to the (arbitrary, but not based on arc length) parameters defined in the profile.
So, the definition in the CPACS file seems to be incosistent (or at least not straightaway visible for the user).

After first reflection, we had the idea to introduce another CPACS child node for the guide curve which has the same basis as the (later used and user defined) parameters. In that way, the user could choose the basis (arc length/parameter based) just by selecting the respective node within the guide curve node.
Another approach we had, is to change the way the parameters are set. As far as I know, they are more or less only introduced for the correct connection of two kinks for lofting. But the actual value does not really matter as long as everything is consistent. That way, the users would not misplace the parameter and the chosen value for the guide curve to start.

What do you think @MarAlder ?

joergbrech, 2024-06-12:
> After first reflection, we had the idea to introduce another CPACS child node for the guide curve which has the same basis as the (later used and user defined) parameters.

Something along the lines of

```xml
<xsd:choice>
    <xsd:element name="fromRelativeCircumference" type="doubleBaseType">
        <xsd:annotation>
            <xsd:documentation>Reference to the relative circumference
                position from which the guide curve shall start. Valid values
                are in the interval -1.0...1.0.
            </xsd:documentation>
        </xsd:annotation>
    </xsd:element>
    <xsd:element name="fromParameter" type="doubleBaseType">
        <xsd:annotation>
            <xsd:documentation>Reference to the parametric position
            from which the guide curve shall start on the profile. The value
            depends on the parametrization of the curve. This can be influenced 
            e.g. via the parameterMap of pointList profiles. This way guide curves
            can be placed at predined kinks.
            </xsd:documentation>
        </xsd:annotation>
    </xsd:element>
</xsd:choice>
```

closed 2024-08-02 by svengoldberg