DLR-SC/tigl #752: Creating a new file in CPACS Creator is not possible because CPACS wants to write to the installation directory

bradphelan, 2020-10-15:
[image]

I select *empty* then

[image]

The software should try to write into the template directory when trying to read a template I think

joergbrech, 2020-10-15:
Thanks for raising an issue here!

When you click on the Settings button (the cogs symbol) you should be able to specify a template directory that you have write access to. Could you check if this resolves your problem?

rainman110, 2020-10-28:
@joergbrech I think we should destinguish between a template directory (which could be readonly) and a temp directory to create a new file.

Imho, the error should not occur, independent where the templates are stored.

joergbrech, 2025-05-22:
Ok, lets revive this issue. 

The problem is that the "New File" dialog creates a temporary file in the templates directory, which might be read-only. Here are two possible solutions:

**Solution A:** We don't create a temporary file at all but copy the contents from the template into a CPACS configuration in RAM. When the user first saves the file, a QFileDialog opens and the user can select a path and file name. 

**Solution B:** If solution A is difficult to implement because of CCPACSConfiguration & TiXI work *(I am not entirely sure right now)* we could fall back to a simpler solution, where a user must select a path and filename directly when clicking "New File".

Slightly unrelated, probably a separate issue: We should keep track of whether  there have been unsaved changes, maybe have a visible marker somewhere and warn before the user closes the CPACSCreator without saving the document.