DLR-SC/tigl #756: Change absolute tolerance values to relative

rainman110, 2020-10-27:
This is a huge issue and must probably be split into multiple smaller ones.

Absolute values have the problem, the the algorithms are not invariant of model scaling and might fail.

Therefore, we should
 1. Find all places, where we use hard tolerances (search for small values, Precision::Confusion)
 2. We need a concept, of how to pass reference sizes (e.g.) of the aircraft, shapes... into the functions
 3. Finally replace absolute values by relative_tolerance * refercence_size

joergbrech, 2020-10-28:
I agree! We should have a consistent definition of tolerances. We should probably also have different relative tolerances for 

 - spatial values (length, area, volume)
 - radians & degrees
 - geometric parameters

> We need a concept, of how to pass reference sizes (e.g.) of the aircraft, shapes... into the functions

Any idea on how we could achieve this? Some algorithms for calculating reference sizes (e.g. the arc length of a curve or a bounding box) are iterative and require a tolerance value themself. If we go about this the wrong way, we may clutter our low-level algorithms with the calculation of reference sizes. We have to make sure that performance doesn't suffer too much by this.

rainman110, 2020-10-28:
> If we go about this the wrong way, we may clutter our low-level algorithms with the calculation of reference sizes. 

The low level algorithms should not have to compute the tolerances, but it should be passed into them. We need to compute absolute tolerances on the CPACS level I would say ( i think it really depends on the algorithm)

To not mix up the tolerances, it would make sense to define separate C++ types for tolerances, as e.g.
 - SpatialTolerance (Absolute and Relative)
 - ParametricTolerance
 - and whatever we need

joergbrech, 2021-03-01:
We should discuss how to handle different units as well