DLR-SC/tigl #761: Tigl 3.1.0 precompiled MEX file

TimmHb, 2020-11-13:
I’m using it on matlab but it seems i’m missing the precompiled MEX file: tigl_matlab.mexw64...which I have in Tigl 2.2.1 and tigl 3.0.0..
I’ve just downloaded the zip file from https://github.com/DLR-SC/tigl/releases/tag/v3.1.0  but it is not in the share/matlab/ folder..
Am I missing something?

joergbrech, 2020-11-13:
No, you are right. Unfortunately the mex files are missing in the 3.1.0 release. We switched our automated release build chain from travis/appveyor to Github actions and somehow the Matlab mex file got lost on the way. For future releases this has been fixed in https://github.com/DLR-SC/tigl/commit/0f564cdddf3c8b72675c7b709ccd17971c56d9c5.

So for now you have two options:
1. If you have to use the 3.1.0 release, you would need to build the mex file yourself. You can use our [python script](https://github.com/DLR-SC/tigl/blob/master/bindings/matlab/make_tigl_matlab.py) for it and oriente yourself using our [CMakeLists.txt file](https://github.com/DLR-SC/tigl/blob/a00eba27266528f67eef3b6e4e8b0c523105ba19/bindings/matlab/CMakeLists.txt#L17). 
2. You can use the latest development version of TiGL, our master branch should be fairly. 

   You can always download the latest build from Github by switching to the "Actions" tab, choosing the newest "Continuous Integration" build and then downloading the artifact that you need. The latest development build for Win64 as of writing can be found [here](https://github.com/DLR-SC/tigl/suites/1495358564/artifacts/26058109). It contains the Matlab mex file.

rainman110, 2020-11-13:
A third option is to compile the mex file on your own using matlab's `mex` tool. 

Inside tigl installation folder `share/tigl3/matlab` you can find a `Makefile`. Ideally, you can simply run 

```bat
nmake
```

In windows to compile it. If you don't have nmake, you can directly call

```
mex tiglmatlab.c -I../../../include/tigl3 -L../../../lib -ltigl3 -output tigl_matlab
```

from inside the folder.

TimmHb, 2020-11-17:
Hi joergbrech and rainman, thnx for commenting. I tried building it with the mex command in matlab but got the following error;
I'm using the  'MinGW64 Compiler (C)'. I can give it a try with python. 

Error using mex
In file included from
\TIGL-3.1.0-win64\TIGL-3.1.0-win64\share\tigl3\matlab\tiglmatlab.c:39:0:
\TIGL-3.1.0-win64\TIGL-3.1.0-win64\share\tigl3\matlab\common.h: In function
'isallinteger':
\TIGL-3.1.0-win64\TIGL-3.1.0-win64\share\tigl3\matlab\common.h:293:10: error:
redeclaration of 'i' with no linkage
   mwSize i;
          ^
TIGL-3.1.0-win64\TIGL-3.1.0-win64\share\tigl3\matlab\common.h:258:16: note:
previous declaration of 'i' was here
         mwSize i;
                ^
In file included from
\SOFTWARE\TIGL-3.1.0-win64\TIGL-3.1.0-win64\share\tigl3\matlab\tiglmatlab.c:142:0:
..\..\..\include\tigl3/tigl.h: At top level:
..\..\..\include\tigl3/tigl.h:30:18: fatal error: tixi.h: No such file or directory
compilation terminated.

rainman110, 2020-11-17:
Hi. The first error is an actual error that was fixed just after the tigl 3.1 release:

https://github.com/DLR-SC/tigl/commit/3e3904df4ff45134f480a0802475cd2ed5a207a3

You just need to delete the line 293 in common.h.

To fix the second problem, just download the tixi 3.1.1 libary and copy the the files
 - include/tixi3/tixi.h 
 - lib/tixi3.lib

into the include / lib directory of your tigl installation folder.

TimmHb, 2020-11-17:
Seems to work now, thank you!

closed 2020-11-24 by rainman110