DLR-SC/tigl #765: Generation of spar segment geometry very slow

hardt-fa-flb, 2020-12-09:
The generation of the geometries of the spars are very slow (about 15 minutes). The example consists of two spars with two segments each.

[IEA-15-240-RWT_CPACS.zip](https://github.com/DLR-SC/tigl/files/5666376/IEA-15-240-RWT_CPACS.zip)

rainman110, 2020-12-09:
I can't reproduce it. It is slow on my laptop, but about ~ 1 minute.

It could help to reduce the number of sample points per profile. You have now about 200 points per profile. Try to reduce it to about 50.

hardt-fa-flb, 2024-09-23:
Unfortunately, I have to reactivate this issue, as it is a critical point in a current project. 

We have reduced the number of airfoil points to 20 per upper and 20 lower side . Nevertheless, the computing time for calculating the spar loft is still enormous, especially when compared with the computing time for the wing loft.

We used the following test script to compare the calculation times:

```python
from __future__ import print_function

from time import perf_counter

from tixi3 import tixi3wrapper
from tigl3 import tigl3wrapper
import tigl3.configuration
import os

def get_lofts(configuration):

    from OCC.Display.SimpleGui import init_display
    display, start_display, add_menu, add_function_to_menu = init_display()
    display.Context.SetDeviationCoefficient(0.0001)

    uid_mgr = configuration.get_uidmanager()
    wing = uid_mgr.get_geometric_component("blade")

    # Get wing loft
    t = perf_counter()
    wing_loft = wing.get_loft().shape()
    print(f'wing.get_loft(): {perf_counter() - t:.2f} s')
    display.DisplayShape(wing_loft, color="blue", transparency=0.5)

    # Get spar lofts
    for spar_uid in ['web0', 'web1']:
        t = perf_counter()
        spar = uid_mgr.get_geometric_component(spar_uid)
        spar_loft = spar.get_loft().shape()
        print(f'{spar_uid}: spar.get_loft(): {perf_counter() - t:.2f} s')
        display.DisplayShape(spar_loft, color="red")

    display.FitAll()
    start_display()

if __name__ == '__main__':
    tixi_h = tixi3wrapper.Tixi3()
    tigl_h = tigl3wrapper.Tigl3()

    dir_path = os.path.dirname(os.path.realpath(__file__))
    tixi_h.open(dir_path + "/IEA-15-240-RWT_20_CS.xml")
    tigl_h.open(tixi_h, "")

    # get the configuration manager
    mgr = tigl3.configuration.CCPACSConfigurationManager_get_instance()

    # get the CPACS configuration, defined by the tigl handle
    # we need to access the underlying tigl handle (that is used in the C/C++ API)
    config = mgr.get_configuration(tigl_h._handle.value)
    get_lofts(config)
```

The CPACS models correspond to the [IEA-15-240-RWT wind turbine](https://github.com/IEAWindTask37/IEA-15-240-RWT), discretized to different degrees across the span: [IEA-15-240-RWT.zip](https://github.com/user-attachments/files/17102664/IEA-15-240-RWT.zip)

The output of the skip for the different CPACS files is as follows:

10 spanwise stations:
```
WRN 09/23 20:43:54 tigl.cpp:198] CPACS dataset version is higher than TIGL library version!
WRN 09/23 20:43:56 CTiglWingProfilePointList.cpp:450] The wing profile airfoil_0.00 will be trimmed to avoid a skewed trailing edge. The lower part is trimmed about 0.648035 % w.r.t. the chord length. Please correct the wing profile!
WRN 09/23 20:43:56 CTiglWingProfilePointList.cpp:450] The wing profile airfoil_0.00 will be trimmed to avoid a skewed trailing edge. The lower part is trimmed about 0.381963 % w.r.t. the chord length. Please correct the wing profile!
WRN 09/23 20:43:56 CTiglWingProfilePointList.cpp:466] The wing profile airfoil_0.00 will be trimmed to avoid a skewed trailing edge. The upper part is trimmed about 0.388684 % w.r.t. the chord length. Please correct the wing profile!
WRN 09/23 20:43:56 CTiglWingProfilePointList.cpp:450] The wing profile airfoil_10.00 will be trimmed to avoid a skewed trailing edge. The lower part is trimmed about 0.0948738 % w.r.t. the chord length. Please correct the wing profile!
[... 474 lines cut ...]
xmldiff                   2.7.0                    pypi_0    pypi
xorg-fixesproto           5.0               hcd874cb_1002    conda-forge
xorg-kbproto              1.0.7             hcd874cb_1002    conda-forge
xorg-libx11               1.8.9                hefa74cf_0    conda-forge
xorg-libxau               1.0.11               hcd874cb_0    conda-forge
xorg-libxcursor           1.2.0                hcd874cb_1    conda-forge
xorg-libxdmcp             1.1.3                hcd874cb_0    conda-forge
xorg-libxext              1.3.4                hcd874cb_2    conda-forge
xorg-libxfixes            5.0.3             hcd874cb_1004    conda-forge
xorg-libxinerama          1.1.5                hfa435f1_0    conda-forge
xorg-libxrender           0.9.11               hcd874cb_0    conda-forge
xorg-renderproto          0.11.1            hcd874cb_1002    conda-forge
xorg-xextproto            7.3.0             hcd874cb_1003    conda-forge
xorg-xproto               7.0.31            hcd874cb_1007    conda-forge
xyzservices               2024.6.0           pyhd8ed1ab_0    conda-forge
xz                        5.2.6                h8d14728_0    conda-forge
yaml                      0.2.5                h8ffe710_2    conda-forge
zarr                      2.18.2                   pypi_0    pypi
zeromq                    4.3.5                he1f189c_4    conda-forge
zict                      3.0.0              pyhd8ed1ab_0    conda-forge
zipp                      3.19.2             pyhd8ed1ab_0    conda-forge
zlib                      1.3.1                h2466b09_1    conda-forge
zstd                      1.5.6                h0ea2cb4_0    conda-forge

```

rmaierl, 2024-09-24:
Good morning,

I also cannot reproduce the issue.
The geometry is built in a few seconds.

The generation of the intersected geometry takes a bit longer (1 minute)
Have you thought about using CST profile definition?
This accelerates the geometry generation on our side.

You could also review the geometry in TiGL directly, to check how long it takes there.

Please find some attached screenshots as additional explanation of my points above.
[image]
[image]

The definition of the material is also not complete:
[image]

Additionally I would reccomend to clean up your CPACS file according to the schema:

`08:48:28 [ERROR] (tigl.cpp:132) [TiXI] /caehome/mare102/exchange/IEA-15-240-RWT_CPACS/IEA-15-240-RWT_CPACS.xml:37: Element 'scaling', attribute 'uID': 'cpacs_vehicles_aircraft_model_wings_wing_sections_section[1]_transformation_scaling' is not a valid value of the atomic type 'xs:ID'.
08:48:28 [ERROR] (tigl.cpp:132) [TiXI] /caehome/mare102/exchange/IEA-15-240-RWT_CPACS/IEA-15-240-RWT_CPACS.xml:37: Element 'rotation', attribute 'uID': 'cpacs_vehicles_aircraft_model_wings_wing_sections_section[1]_transformation_rotation' is not a valid value of the atomic type 'xs:ID'.
08:48:28 [ERROR] (tigl.cpp:132) [TiXI] /caehome/mare102/exchange/IEA-15-240-RWT_CPACS/IEA-15-240-RWT_CPACS.xml:37: Element 'translation', attribute 'uID': 'cpacs_vehicles_aircraft_model_wings_wing_sections_section[1]_transformation_translation' is not a valid value of the atomic type 'xs:ID'.
08:48:28 [ERROR] (tigl.cpp:132) [TiXI] /caehome/mare102/exchange/IEA-15-240-RWT_CPACS/IEA-15-240-RWT_CPACS.xml:37: Element 'scaling', attribute 'uID': 'cpacs_vehicles_aircraft_model_wings_wing_sections_section[1]_elements_element_transformation_scaling' is not a valid value of the atomic type 'xs:ID'.
08:48:28 [ERROR] (tigl.cpp:132) [TiXI] /caehome/mare102/exchange/IEA-15-240-RWT_CPACS/IEA-15-240-RWT_CPACS.xml:37: Element 'rotation', attribute 'uID': 'cpacs_vehicles_aircraft_model_wings_wing_sections_section[1]_elements_element_transformation_rotation' is not a valid value of the atomic type 'xs:ID'.
08:48:28 [ERROR] (tigl.cpp:132) [TiXI] /caehome/mare102/exchange/IEA-15-240-RWT_CPACS/IEA-15-240-RWT_CPACS.xml:37: Element 'translation', attribute 'uID': 'cpacs_vehicles_aircraft_model_wings_wing_sections_section[1]_elements_element_transformation_translation' is not a valid value of the atomic type 'xs:ID'.
08:48:28 [ERROR] (tigl.cpp:132) [TiXI] /caehome/mare102/exchange/IEA-15-240-RWT_CPACS/IEA-15-240-RWT_CPACS.xml:37: Element 'scaling', attribute 'uID': 'cpacs_vehicles_aircraft_model_wings_wing_sections_section[2]_transformation_scaling' is not a valid value of the atomic type 'xs:ID'.
08:48:28 [ERROR] (tigl.cpp:132) [TiXI] /caehome/mare102/exchange/IEA-15-240-RWT_CPACS/IEA-15-240-RWT_CPACS.xml:37: Element 'rotation', attribute 'uID': 'cpacs_vehicles_aircraft_model_wings_wing_sections_section[2]_transformation_rotation' is not a valid value of the atomic type 'xs:ID'.
08:48:28 [ERROR] (tigl.cpp:132) [TiXI] /caehome/mare102/exchange/IEA-15-240-RWT_CPACS/IEA-15-240-RWT_CPACS.xml:37: Element 'translation', attribute 'uID': 'cpacs_vehicles_aircraft_model_wings_wing_sections_section[2]_transformation_translation' is not a valid value of the atomic type 'xs:ID'.
08:48:28 [ERROR] (tigl.cpp:132) [TiXI] /caehome/mare102/exchange/IEA-15-240-RWT_CPACS/IEA-15-240-RWT_CPACS.xml:37: Element 'scaling', attribute 'uID': 'cpacs_vehicles_aircraft_model_wings_wing_sections_section[2]_elements_element_transformation_scaling' is not a valid value of the atomic type 'xs:ID'.
08:48:28 [ERROR] (tigl.cpp:132) [TiXI] /caehome/mare102/exchange/IEA-15-240-RWT_CPACS/IEA-15-240-RWT_CPACS.xml:37: Element 'rotation', attribute 'uID': 'cpacs_vehicles_aircraft_model_wings_wing_sections_section[2]_elements_element_transformation_rotation' is not a valid value of the atomic type 'xs:ID'.
08:48:28 [ERROR] (tigl.cpp:132) [TiXI] /caehome/mare102/exchange/IEA-15-240-RWT_CPACS/IEA-15-240-RWT_CPACS.xml:37: Element 'translation', attribute 'uID': 'cpacs_vehicles_aircraft_model_wings_wing_sections_section[2]_elements_element_transformation_translation' is not a valid value of the atomic type 'xs:ID'.
08:48:28 [ERROR] (tigl.cpp:132) [TiXI] /caehome/mare102/exchange/IEA-15-240-RWT_CPACS/IEA-15-240-RWT_CPACS.xml:37: Element 'scaling', attribute 'uID': 'cpacs_vehicles_aircraft_model_wings_wing_sections_section[3]_transformation_scaling' is not a valid value of the atomic type 'xs:ID'.
08:48:28 [ERROR] (tigl.cpp:132) [TiXI] /caehome/mare102/exchange/IEA-15-240-RWT_CPACS/IEA-15-240-RWT_CPACS.xml:37: Element 'rotation', attribute 'uID': 'cpacs_vehicles_aircraft_model_wings_wing_sections_section[3]_transformation_rotation' is not a valid value of the atomic type 'xs:ID'.
08:48:28 [ERROR] (tigl.cpp:132) [TiXI] /caehome/mare102/exchange/IEA-15-240-RWT_CPACS/IEA-15-240-RWT_CPACS.xml:37: Element 'translation', attribute 'uID': 'cpacs_vehicles_aircraft_model_wings_wing_sections_section[3]_transformation_translation' is not a valid value of the atomic type 'xs:ID'.
08:48:28 [ERROR] (tigl.cpp:132) [TiXI] /caehome/mare102/exchange/IEA-15-240-RWT_CPACS/IEA-15-240-RWT_CPACS.xml:37: Element 'scaling', attribute 'uID': 'cpacs_vehicles_aircraft_model_wings_wing_sections_section[3]_elements_element_transformation_scaling' is not a valid value of the atomic type 'xs:ID'.
[... cut ...]
08:48:38 [ERROR] (tigl.cpp:132) [TiXI] /caehome/mare102/exchange/IEA-15-240-RWT_CPACS/IEA-15-240-RWT_CPACS.xml:438: Element 'nodalLoads': This element is not expected. Expected is one of ( description, specification, loadData ).
08:48:38 [ERROR] (tigl.cpp:132) [TiXI] /caehome/mare102/exchange/IEA-15-240-RWT_CPACS/IEA-15-240-RWT_CPACS.xml:451: Element 'nodalLoads': This element is not expected. Expected is one of ( description, specification, loadData ).
08:48:38 [ERROR] (tigl.cpp:132) [TiXI] /caehome/mare102/exchange/IEA-15-240-RWT_CPACS/IEA-15-240-RWT_CPACS.xml:464: Element 'nodalLoads': This element is not expected. Expected is one of ( description, specification, loadData ).`

ewerthen, 2024-09-24:
Hi,
thank you for your fast response. We will figure out the CST profile definition. So the generation of the intersected geometry for the file "IEA-15-240-RWT_30_CS.xml" given above works for you without errors? That would be strange. The material with the uid "glass_triax" is given in all CPACS files:
[image]

rmaierl, 2024-10-10:
Please find the intersected geometry here:

You also need to add the material (or composite) UId to the structural component.
Then the material properties from the material node are taken into account for this element.

[IEA-15-240-RWT_intersected_stp.zip](https://github.com/user-attachments/files/17330630/IEA-15-240-RWT_intersected_stp.zip)

[image]
[image]

ewerthen, 2024-10-17:
Thank you for our response. For us it is not clear what you mean by “material uid to the structural component”? From our understanding everywhere (where it is required by CPACS) the material is set. Even if this is the case, the CAD geometry intersection should still work just an error should occur that a material has not been set