DLR-SC/tigl #779: Problem showing an aeircraft with engine, but without nacelle

CLiersch, 2021-03-15:
When just applying an engine to an aircraft (here called "bothEngines") without a nacelle, TiGL-opens an error dialog box in the GUI:
[image]

After clicking on "close", TiGL proceeds and shows the rest of the geometry - but from my perspective it should be fine to have an aircraft with engine but without nacelle (e.g. engine buried in the fuselage).

rainman110, 2021-03-22:
I agree it should be possible. What does the CPACS standard say? Is a nacelle mandatory or optional?

MarAlder, 2021-03-22:
It's optional: [engineType](https://www.cpacs.de/documentation/CPACS_3_2_0_Docs/html/83f3d6ba-00ff-fcde-5286-37eaeac6088e.htm)

joergbrech, 2021-04-30:
I agree that this error message is wrong and should be fixed. In the mean time, starting with TiGL 3.2, you can suppress error messages in TiGLViewer with the command line option `--suppress-errors`.

joergbrech, 2022-06-15:
If I remember correctly, the problem was that we try to build a tree of `RelativelyPositionedComponent`s. Therefore the `CPACSEnginePosition` was initially implemented as a `RelativelyPositionedComponent`, ignoring the fact that it does not necessarily contain any geometric information.

joergbrech, 2022-06-15:
The same problem occurs with pylons, can be reproduced e.g. with https://github.com/DLR-SC/tigl/blob/master/tests/unittests/TestData/CPACS_30_D250_10.xml.