DLR-SC/tigl #793: Regression: Cannot build loft of Ariane 5

joergbrech, 2021-04-23:
TiGL Viewer crashes on the current master with a vector subscript out of range caused by this line: 

https://github.com/DLR-SC/tigl/blob/7320a3c8531a332b6cfbd0c64940b5278e1252ff/src/fuselage/CTiglFuselageSegmentGuidecurveBuilder.cpp#L97

when opening the [Ariane 5](https://github.com/DLR-SC/tigl-examples/tree/master/cpacs#ariane-5) with tiglviewer.

rainman110, 2021-04-23:
Do you have any clue, what causes the bug /when the bug was introduced?

rainman110, 2021-04-23:
At least cool, that this was detected by your script now ;)

joergbrech, 2021-04-23:
You were faster than me, I am still git bisecting. 

:D I like that the commit message is "prevent hard crash ..."

closed 2021-04-23 by rainman110

rainman110, 2021-04-23:
> I like that the commit message is "prevent hard crash ..." 😆

Yeah, lets just replace one crash with another 😆

PR #795 (2021-04-23): hotfix for #767

Fixes #793

Fixes a small bug introduced in https://github.com/DLR-SC/tigl/commit/2164b67f8fe7857f179c111caf6e804dbdc836d9 where I check for the first and last element of a vector that is allowed to be empty.

## Description

## How Has This Been Tested?

## Screenshots, that help to understand the changes(if applicable):

## Checklist:

- [ ] A test for the new functionality was added.
- [ ] All tests run without failure.
- [ ] The new code complies with the TiGL style guide.
- [ ] New classes have been added to the Python interface.
- [ ] API changes were documented properly [... cut]