DLR-SC/tigl #810: tigl3 and tigl2 compatibility

sfreund-DLR, 2021-06-16:
Hi,

with the newest tigl version in the Python ecosystem (conda-forge) tigl3 and tigl(2) have conflicting requirements. Thus it is not possible anymore to have tigl3 and tigl2 installed in the same python distro and it is not possible to load cpacs2 and cpacs3 files during the same run. Is there a simple way to make these versions compatible again?

- tigl2 is only compatible to python3.7 or less
- tigl3 has dependencies, that require python3.8

**Requirements file:**
> name: py37x
> channels:
>   - dlr-sc
>   - conda-forge
> dependencies:
>   - python=3.7
>   - tigl3=3.2.0
>   - tigl

**Conda output:**
> $ conda env create -f requirements_py37_miniforge2.yml
> Collecting package metadata (repodata.json): ...working... done
> 
> Found conflicts! Looking for incompatible packages.
> This can take several minutes.  Press CTRL-C to abort.
> failed
> 
> Found conflicts! Looking for incompatible packages.
> This can take several minutes.  Press CTRL-C to abort.
> failed
> 
> UnsatisfiableError: The following specifications were found to be incompatible with each other:
> 
> Output in format: Requested package -> Available versions
> 
> Package python conflicts for:
> python=3.7
> tigl3=3.2.0 -> pyqt -> python[version='2.7.*|3.5.*|3.6.*|>=2.7,<2.8.0a0|>=3.9,<3.10.0a0|>=3.5,<3.6.0a0|3.4.*']
> tigl3=3.2.0 -> python[version='>=3.6,<3.7.0a0|>=3.7,<3.8.0a0|>=3.8,<3.9.0a0']
> 
> Package pyqtchart conflicts for:
> tigl3=3.2.0 -> pyqt -> pyqtchart==5.12[build='py39h415ef7b_5|py38h885f38d_6|py39h415ef7b_6|py37hf2a7229_6|py36he2d232f_7|py39h415ef7b_7|py38h885f38d_7|py37hf2a7229_7|py36he2d232f_6']
> tigl -> pyqt -> pyqtchart==5.12[build='py39h415ef7b_5|py38h885f38d_6|py39h415ef7b_6|py37hf2a7229_6|py36he2d232f_7|py39h415ef7b_7|py38h885f38d_7|py37hf2a7229_7|py36he2d232f_6']

rainman110, 2021-06-16:
Oh really? This should not be the case (in theory), but let me check. We don't use any conda-forge dependencies for our build process.

One possible problem is, that the DLR switched to miniforge, which uses conda-forge as a default repo. Maybe this causes package conflicts.

rainman110, 2021-06-16:
tigl2 should be available for python 3.5, 3.6 and 3.7
tigl3 should be available for pythob 3.6, 3.7 and 3.8

It seems that somehow pyqt is causing the problems.

rainman110, 2021-06-16:
Lets have a look at the depenencies:

`conda create -n test_test python=3.7 tigl3=3.2.0 -c dlr-sc `

results in

```
  ca-certificates    pkgs/main/win-64::ca-certificates-2021.5.25-haa95532_1
  certifi            pkgs/main/win-64::certifi-2021.5.30-py37haa95532_0
  freeimageplus      dlr-sc/win-64::freeimageplus-3.18.0-h6538335_2
  freetype           dlr-sc/win-64::freetype-2.6.5-hfa6e2cd_4
  icu                pkgs/main/win-64::icu-58.2-ha925a31_3
  jpeg               pkgs/main/win-64::jpeg-9b-hb83a4c4_2
  libpng             pkgs/main/win-64::libpng-1.6.37-h2a8f88b_0
  opencascade        dlr-sc/win-64::opencascade-7.4.0-h7e01d4e_7
  openssl            pkgs/main/win-64::openssl-1.1.1k-h2bbff1b_0
  pip                pkgs/main/win-64::pip-21.1.2-py37haa95532_0
  pyqt               pkgs/main/win-64::pyqt-5.9.2-py37h6538335_2
  python             pkgs/main/win-64::python-3.7.10-h6244533_0
  pythonocc-core     dlr-sc/win-64::pythonocc-core-7.4.1-py37h7e01d4e_3
  qt                 pkgs/main/win-64::qt-5.9.7-vc14h73c81de_0
  setuptools         pkgs/main/win-64::setuptools-52.0.0-py37haa95532_0
  sip                pkgs/main/win-64::sip-4.19.8-py37h6538335_0
  six                pkgs/main/noarch::six-1.16.0-pyhd3eb1b0_0
  sqlite             pkgs/main/win-64::sqlite-3.35.4-h2bbff1b_0
  tbb                dlr-sc/win-64::tbb-2019.5-h7e01d4e_1
  tigl3              dlr-sc/win-64::tigl3-3.2.0-py37h7e01d4e_1
  tixi3              dlr-sc/win-64::tixi3-3.1.1-py37h9d197f5_2
  vc                 pkgs/main/win-64::vc-14.2-h21ff451_1
  vs2015_runtime     pkgs/main/win-64::vs2015_runtime-14.27.29016-h5e58377_2
  wheel              pkgs/main/noarch::wheel-0.36.2-pyhd3eb1b0_0
  wincertstore       pkgs/main/win-64::wincertstore-0.2-py37_0
  zlib               pkgs/main/win-64::zlib-1.2.11-h62dcd97_4
```

rainman110, 2021-06-16:
`conda create -n test_test python=3.7 tigl -c dlr-sc`

results in 

``` 
  ca-certificates    pkgs/main/win-64::ca-certificates-2021.5.25-haa95532_1
  certifi            pkgs/main/win-64::certifi-2021.5.30-py37haa95532_0
  freeimageplus      dlr-sc/win-64::freeimageplus-3.18.0-h6538335_2
  freetype           dlr-sc/win-64::freetype-2.6.5-hfa6e2cd_4
  gl2ps              dlr-sc/win-64::gl2ps-1.3.8-hfa6e2cd_4
  icu                pkgs/main/win-64::icu-58.2-ha925a31_3
  jpeg               pkgs/main/win-64::jpeg-9b-hb83a4c4_2
  libpng             pkgs/main/win-64::libpng-1.6.37-h2a8f88b_0
  oce                dlr-sc/win-64::oce-0.17.2-he980bc4_14
  openssl            pkgs/main/win-64::openssl-1.1.1k-h2bbff1b_0
  pip                pkgs/main/win-64::pip-21.1.2-py37haa95532_0
  pyqt               pkgs/main/win-64::pyqt-5.9.2-py37h6538335_2
  python             pkgs/main/win-64::python-3.7.10-h6244533_0
  pythonocc-core     dlr-sc/win-64::pythonocc-core-0.17.3-py37he980bc4_10
  qt                 pkgs/main/win-64::qt-5.9.7-vc14h73c81de_0
  setuptools         pkgs/main/win-64::setuptools-52.0.0-py37haa95532_0
  sip                pkgs/main/win-64::sip-4.19.8-py37h6538335_0
  sqlite             pkgs/main/win-64::sqlite-3.35.4-h2bbff1b_0
  tbb                dlr-sc/win-64::tbb-2019.5-h7e01d4e_1
  tigl               dlr-sc/win-64::tigl-2.2.3-py37he980bc4_3
  tixi               dlr-sc/win-64::tixi-2.2.4-vc14_1
  vc                 pkgs/main/win-64::vc-14.2-h21ff451_1
  vs2015_runtime     pkgs/main/win-64::vs2015_runtime-14.27.29016-h5e58377_2
  wheel              pkgs/main/noarch::wheel-0.36.2-pyhd3eb1b0_0
  wincertstore       pkgs/main/win-64::wincertstore-0.2-py37_0
  zlib               pkgs/main/win-64::zlib-1.2.11-h62dcd97_4

```

rainman110, 2021-06-16:
All packages are identical except from six, tigl / tigl3, tixi / tixi3 and oce / opencascade.

rainman110, 2021-06-28:
I think the main problems are the different pythonocc-core version required. TiGL 3.2 requires pythonOCC=7.4.x, TiGL 2.X and TiGL < 3.2 requires pythonOCC=0.17.x. Hence , it is not possible right now to install both packages in the same environment due to a conflict.

I need to check, whether we can repackage cpacs2to3, such that we don't need both TiGL packages (but just the TiGL 2.x DLL for example).

rainman110, 2021-06-30:
@sfreund-DLR I created a workaround to make it possible for a parallel tigl2 and tigl3 installations:

I created a new package tigl-static (TiGL 2) which statically links OCE 0.17 such that we don't collide with OpenCASCADE dependencies. The downside is, that the tigl-static package only has tiglwrapper.py, not the bindings of the C++ API.

An alternative solution is to stick with TIGl 3.1, which uses the same OCE version as TiGL 2.

I also updated the cpacs2to3 package, that now uses the tigl-static package.

closed 2021-07-02 by rainman110

sfreund-DLR, 2021-07-02:
I just tested it. Everything works fine - thanks a lot!