DLR-SC/tigl #826: Unit / integration tests for python bindings.

rainman110, 2021-08-11:
We need test for the python bindings. As at least on of the CI jobs builds the bindings, we can test them easily.

MarAlder, 2024-11-28:
From #1012:

> > Perhaps we should strive to also cover such Python functions via unittests?
> 
> I totally agree, but I wouldn't want to add it to this issue/PR-pair.
> 
> Any idea on a pragmatic way to achieve this? Here are some ramblings:
> 
> - Regardless on how we implement the Python tests, having them in CI would be an improvement.
Yep

> - When we implement the python tests, we could agree that we don't actually have to test the functionality of the swig-wrapped classes in the Python unit tests, but that it suffices to just the existence of the classes in the Python interface. This would limit the maintenance overhead of TiGL. It might be possible to achieve this with some degree of automation, not sure. Without automation, every contributor must add these checks themselves and the reviewer must manually verify that all relevant checks are in place, which adds a (small) burden to the maintenance.

Hmm, I was thinking in the direction that new high-level methods (what we define in CCPACS classes) are covered in separate unittest files (according to how it makes sense thematically, e.g. for ducts, [fuelTanks](https://github.com/MarAlder/tigl/blob/tanks/tests/unittests/test_fuelTanks.py), etc.). This should be somewhat consistent to the C++ unittests. As @rainman110 sais, it is then easy to trigger all tests via CI. 

Automating it 🤔, always good. Not sure about the effort, though. However, it could ensure better coverage of the tests and prevent a patchwork of overlapping or missing tests according to my approach above.

> - We try to derive C++ and Python unit tests from a model-based testing approach. This would involve a lot of work to set the system up and the outcome is not clear: It could be an awesome new thing that boosts robustness or it could turn out to be too complex and impracticable, because it is really hard to formally model the process of modelling with TiGL/CPACS.

Fancy, but perhaps a little over the top? 😏

rainman110, 2024-11-28:
The problem is that we won't have python tests for files that we have not wrapped, since we have forgotten these classes to integrate.

Instead of having a whitelist of what to test, we could do it the other way around. From the cpacs codegen or the existing cpacs classes (header files), we can derive, which python classes need to exist. A black list could exclude classes that we don't want to test.

Then, a python test "simply" checks for all classes that need to exist and tries to instantiate or simply imports them, which should be sufficient for an existence test.

joergbrech, 2024-11-28:
> Instead of having a whitelist of what to test, we could do it the other way around. From the cpacs codegen or the existing cpacs classes (header files), we can derive, which python classes need to exist. A black list could exclude classes that we don't want to test.

Yes, that is what I meant with automation. This requires a bit of code analysis, e.g. identifying inputs and outputs of (member) functions, identifying `std::vector<XXX>` and `std::unique_ptr<XXX>` both through `auto` or `typedef`s. Maybe we could use libclang or something.

> Fancy, but perhaps a little over the top? 😏

Yes, let's not start with that. :)

joergbrech, 2025-09-25:
If #1185 gets merged, it would be a good start to have some manual written python tests in CI. It should be straightforward with a `pythontests` pixi task in the python-internal environment.

While we are at it, I suggest moving the `TestData` directories to a centralized location, so that it can be shared by integration tests, unittests and python tests.

```
tests/
├─ unittests/
├─ integrationtests/
├─ pythontests/
├─ TestData/
```