DLR-SC/tigl #827: Wing aspect ratio value seems not correct

aidanjungo, 2021-09-06:
https://github.com/DLR-SC/tigl/blob/59ae2512e866a69905358d479df91e548fd0b38c/src/wing/CCPACSWing.cpp#L661-L667

I am not sure I completely understand how the formula to calculate AR has been changed recently, but when I use 'get_aspect_ratio()' for wing 1 of the D150 i got 526.71 before this change the value was 9.4.

```
>>> aircraft_tigl.get_wing(1).get_aspect_ratio()
526.7133945257066
```

It seems to me that it use the wrong reference area for the wing in the AR formula

If we use:

AR = (Span**2)/4 / RefArea *2  

Before I think it was:
```
>>> 33.91522073785017*33.91522073785017/4/61.162786877005146*2
9.40312121494867

```

But now it seems to be:
```
>>> 33.91522073785017*33.91522073785017/4.3676208338421665*2
526.7133945257066
```

aidanjungo, 2021-09-06:
I just checked, if I change "x-z-plane" by "x-y-plane" for the symmetry of wing 1, it gives the correct value, so it seems that "get_aspect_ratio()" did not use the correct reference area projection to calculate the AR.

rainman110, 2021-09-06:
Hi @aidanjungo . Thanks for pointing this out. I'll have a look, why this has changed. Which TiGL version have you been using before (which didn't have this bug)?

aidanjungo, 2021-09-07:
I am not sure of which version I was using because I installed cpacscreator via Conda. But now, I feel that Tigl3 has been update but maybe not cpacscreator, is it correct?
I saw the function "get_half_span" has been replace by "get_wingspan" in Tigl3, but it seems it does not exist (yet) in cpacscreator.

rainman110, 2021-09-07:
Indeed, we have not updated cpacs creator yet. It still uses tigl 3.0.0 as a basis.

rainman110, 2021-11-09:
This test demonstrates the problem:
https://github.com/DLR-SC/tigl/runs/4152673156?check_suite_focus=true#step:25:1210

rainman110, 2021-11-09:
Apparently, Malo fixed the bug back then in the cpacs creator. I'll backport the fix into tigl.

commit fd440535 (2021-11-09): Fixed computation of aspect ration

I backported the code from cpacs creator,
which already solved the problem.

Fixes #827
commit 7f265876 (2021-11-10): Fixed failing test

Added copyright header to winghelpers.

Closes #827
commit 238ad41e (2021-11-12): Fixed computation of aspect ration

I backported the code from cpacs creator,
which already solved the problem.

Fixes #827
commit b1af7931 (2021-11-12): Fixed failing test

Added copyright header to winghelpers.

Closes #827
closed 2021-11-12 by rainman110