DLR-SC/tigl #830: CPACS files with patch version are not processed

MarAlder, 2021-09-27:
If we have a CPACS version with patch number, the dataset will not open:
```XML
<header>
    <cpacsVersion>3.3.1</cpacsVersion>
</header>
```

[basicWing.zip](https://github.com/DLR-SC/tigl/files/7236800/basicWing.zip)

[image]

[image]

rainman110, 2021-09-27:
The problem does not only exist in the cpacs schema (which was actually fixed for tigl) but also here in the tigl code:

https://github.com/DLR-SC/tigl/blob/master/src/api/tigl.cpp#L166

MarAlder, 2021-11-02:
Nice fix. Would you suggest to restrict the `stringBaseType` used for `cpacsVersion` in `cpacs_schema.xsd` with the same regex? Could make sense, right?

`([0-9]+)\\.([0-9]+)(\\.([0-9]+))?(?:-(\\w+(\\.\\w*)?)(\\+(.*))?|\\+(.*))?`

commit ee7e3440 (2021-11-02): Fixed parsing the cpacs version

Fixes #830
commit 8cbafe1e (2021-11-02): Fixed parsing the cpacs version

Fixes #830
commit 0605077d (2021-11-04): Clarify ignoring build metadata

Closes #830
PR #847 (2021-11-04): Fixed parsing of CPACS versions

Fixed parsing the cpacs version.

## Description

Previously, the cpacs version could not be parsed correctly, if a patch level version was included in the version string.

This PR adds proper semantic versioning. A new class for parsing semantic versions based on std::regex has been added. Tests have been added accordingly.

Fixes #830 

## How Has This Been Tested?

I added tests for the new semantic versioning class and tests that check opening the cpacs files with correct and incorrect cpacs versions.

## Checklist:

- [x] A test for the new functionality w [... cut]
closed 2021-11-04 by rainman110