DLR-SC/tigl #837: Consider building conda packages with every merge commit on main.

joergbrech, 2021-10-04:
This way, python users can easily download dev versions of TiGL from the build artifacts without having to wait for a new release.

rainman110, 2021-10-04:
Good idea. We need to think about some issue though
 - What about versioning? How do we distinguish stable releases from nightly builds?
 - Currently, the conda package repo is separate from the tigl repo. This means, that we either need to copy over the conda recipe to the tigl repo, or we checkout the recipe during the build process.

joergbrech, 2021-10-06:
Suggestion: 
  - create a workflow on DLR-SC/tigl-conda that can be triggered with workflow_dispatch and some inputs
  - This workflow seds the version string and git-tag in the tigl3 recipe and builds the conda package
  - Use https://github.com/oprypin/nightly.link to get a persistent link to the build artifact of the tigl-conda workflow
  - Use https://github.com/marketplace/actions/workflow-dispatch in this workflow to trigger the tigl-conda build
  - Create a Wiki entry explaining how to download and install development conda-packages.