DLR-SC/tigl #853: IGES/Szep Export fails, if one component fails

rainman110, 2021-11-15:
If we try to export an aircraft, where one component is failing to build, the whole export fails.

It would be better, if the exporter exports everything that can be exported.

@joergbrech What is your opinion? If we fix it, we don't necessarily know that there was an issue. On the other hand, the users succeeds at least a bit creating an export.

joergbrech, 2021-11-16:
Hmm, good question. I guess it would be really convenient if we could postpone all error messages to after the file gets written, but I suppose this would be tricky to do.

rainman110, 2021-11-16:
I currently rewrite the code to figure out, whether there was an issue while exporting. The exporter still exports as much as possible, but an error is returned to the user to notify success or failure.

commit 379be294 (2021-11-17): Improved robustness of the CAD export

If one component of the cpacs model failed
to compute, the whole export failed.  The code
change tries to export whatever it can.

The export function still returns an error, but there
will be a IGES/STEP file the user can work with.

Fixes #853
closed 2021-11-17 by joergbrech