DLR-SC/tigl #859: Windows CI broken

joergbrech, 2021-12-07:
The nightly builds have been failing for a few days now: https://github.com/DLR-SC/tigl/actions
`PermissionError: [Errno 13] Permission denied: 'C:\\Miniconda\\Library\\include\\opencascade'`

rainman110, 2021-12-07:
Is it possible, that the cache (used by clcache) is exausted, i.e. the disk space is full?

rainman110, 2021-12-07:
Can we inspect our current cache size?

joergbrech, 2021-12-08:
> Can we inspect our current cache size?

I don't know...but updating to `actions/cache@v2` did not solve the problem.

rainman110, 2021-12-13:
I could fix windows CI by disabling clcache. I don't know, why it suddenly stopped working. Since the move to github action, we probably don't need clcache anymore as we have parallel builds. Also, the cache seemed to be mostly empty.

commit 2781b653 (2021-12-13): Try to fix #859
commit 0665df57 (2021-12-13): Try to fix #859
closed 2021-12-13 by rainman110