DLR-SC/tigl #868: Installation instructions: OCE up to date?

MarAlder, 2021-12-15:
Moin,

is the [installation instruction](https://github.com/DLR-SC/tigl/wiki/Building-TiGL-from-Source) still up-to date, as it provides the pre-compiled [OCE MSVC2015 binaries](https://sourceforge.net/projects/tigl/files/Thirdparty/oce.0.17.2.win64_msvc2015_static.7z/download) TiGL now requires Open Cascade 7.4?

rainman110, 2021-12-15:
TiGL officially uses now OpenCASCADE 7.4.0 but it should be still possible to build with OCE 0.17.2

rainman110, 2021-12-15:
We need to update the installation instructions though

MarAlder, 2021-12-15:
Ah true, my CMake just jumped into the [else conditon in UseOpenCASCADE](https://github.com/DLR-SC/tigl/blob/e3a80e5855a8dd1e09ca84f8d888ab6f93464bd7/cmake/UseOpenCASCADE.cmake#L49) because I confused again backward slash with forward slash for `-DCASROOT`... 😏 Now it correctly includes OCE 0.17.2.

closed 2021-12-15 by MarAlder