DLR-SC/tigl #869: False positives in check, if wing cells can be trimmed rather than cut.

joergbrech, 2021-12-21:
TiGL tries to determine if the geometry of a wing cell can be calculated by trimming a surface rather than cutting it with Boolean operations for a better performance. Currently, this check often has false positives and the trimmed result is wrong. The trimming is currently disabled in https://github.com/DLR-SC/tigl/commit/da2d269c6e943f7b9a2376e10babe2f36b56469d and we should either remove the check completely or change it to be more reliable.

This is related to https://github.com/DLR-SC/tigl/commit/da2d269c6e943f7b9a2376e10babe2f36b56469d and #865.

condoratberlin, 2022-03-08:
Currently I would prefer to remove the checks, because the current version leads to wrong cell shapes.

joergbrech, 2022-03-08:
@condoratberlin thanks for the feedback. I will prepare a patch release of TiGL either this or next week where the checks are disabled. I will keep this issue open for documentation purposes. If there are no further issues with the disabled tests, we can remove the checks completely at a later stage (which would just be a refactoring of the source code, without any behavioral changes).