DLR-SC/tigl #875: Ubuntu binaries: error while loading shared libraries: libfreeimage.so.3

iiwolf, 2022-02-03:
Hello, I'm on Ubuntu 18.04 and I downloaded the binary from:

https://github.com/DLR-SC/tigl/releases/download/v3.2.2/TIGL-3.2.2-ubuntu_18.04-x86_64.tar.gz

When I tried to run the binary I got the following error.

```
error while loading shared libraries: libfreeimage.so.3: cannot open shared object file: No such file or directory
```

Let me know if you need more information! Thanks!

iiwolf, 2022-02-03:
As expected, this solved the issue

```
sudo apt-get install libfreeimage-dev                           
```

But I didn't see requirements anywhere and though the binaries included everything required.

joergbrech, 2022-02-04:
Thanks for opening an issue here, @iiwolf!

The recommended way to install TiGL on Linux is by following the instructions in our wiki: https://github.com/DLR-SC/tigl/wiki/Downloads#linux. Here you will find a link to the OpenBuildService with installation instructions for your distribution, or - if you restricted rights - you can use the provided shell script. This will make sure all requirements are met.

Another option would be to install tigl using Anaconda/Miniconda/Miniforge, though this installation only includes the library and not the TiGL Viewer:

```
conda install tigl3 -c dlr-sc
```

rainman110, 2022-02-04:
I think what @iiwolf  is trying to say is, that freeimage is probably not downloaded via the get_tixi_tigl.sh script. Otherwise, it would be part of the ubuntu package (i suppose)