DLR-SC/tigl #881: implement ducts

AntonReiswich, 2022-03-07:
- [ ] add CPACS schema to TiGL repo https://github.com/DLR-SL/CPACS/tree/ducts 
 - [ ] generate code with code generator
 - [ ] define test geometries and unit tests
 - [ ] implement geometry

MarAlder, 2022-04-06:
I see you are already busy implementing the ducts. We recently worked together on [this proposal](https://github.com/DLR-SL/CPACS/tree/ducts) for a corresponding scheme:
[image]

It's just going through my mind: We wanted to create multiple `duct` elements like a `fuselage`, then fuse them and cut the resulting `ductAssembly` out the parent structure. Does this cause problems when defining the inner structure (ribs and spars)? And what happens if the individual `duct` elements don't overlap/connect?

From a CPACS point of view we have the flexibility to use multiple elements within a section. This could also be used to define multiple segments reusing the same elements: 

[image]

```xml
<segment>
    <name>segment1</name>
    <fromElementUID>sec1Element1</fromElementUID>
    <toElementUID>sec2Element1</toElementUID>
</segment>
<segment>
    <name>segment2</name>
    <fromElementUID>sec1Element2</fromElementUID>
    <toElementUID>sec2Element1</toElementUID>
</segment>
<segment>
    <name>segment3</name>
    <fromElementUID>sec2Element1</fromElementUID>
    <toElementUID>sec3Element1</toElementUID>
</segment>
```

The question is certainly how to deal with tangent continuity in Section2, so I intentionally did not remove the two overlapping curves in segment3 for once. A challenge would also be that it drastically changes the lofting algorithm, I guess. 

Anyway, I just want to share these thoughts. Maybe it will inspire one or two ideas for a solution.

joergbrech, 2022-04-06:
Thanks @MarAlder for your comments and help pushing this definition forward!

> Does this cause problems when defining the inner structure (ribs and spars)?

Yes, probably. We haven't started implementing the duct structure yet, so I can't say for certain. We could argue, that the author of the CPACS file is responsible for this: If they define overlapping ducts, they should have an idea how this effects the inner structure.

> And what happens if the individual duct elements don't overlap/connect?

In this case, a `DuctAssembly` just represents a collection of nonoverlapping duct elements. I think this could also be useful to define sets of ducts for individual components. From an implementation perspective, this is no problem.

> The question is certainly how to deal with tangent continuity in Section2, so I intentionally did not remove the two overlapping curves in segment3 for once.

The nice thing is, that this already works by defining dummy guide curves with the `continuity` flag. We can actually prescribe the tangents and continuity conditions at this section and our current unit tests already uses this functionality: 

[image]

Here, the "Y-Duct" assembly in red is defined as a single duct element with symmetry flag and guide curves to describe the tangents at the second section. See the `tangent` and `continuity` nodes:

https://github.com/DLR-SC/tigl/blob/6c8d046acfb6b5d00bb8c5dc29fda65081c0dc62/tests/unittests/TestData/simpletest_ducts.xml#L1651-L1702

Finally, do you think we should open an issue over at [DLR-SL/CPACS](https://github.com/DLR-SL/CPACS) so that the community can comment on the current proposal?

closed 2022-05-13 by joergbrech

rainman110, 2022-05-13:
Awesome work guys 😄

PR #889 (2022-05-13): Implement ducts

## Description

Fixes #881. This PR adds the geometric implementation for ducts.

## How Has This Been Tested?

New unit tests have been added.

## Screenshots, that help to understand the changes(if applicable):

**To Do**

## Checklist:

- [x] A test for the new functionality was added.
- [x] All tests run without failure.
- [x] The new code complies with the TiGL style guide.
- [x] New classes have been added to the Python interface.
- [x] API changes were documented properly in tigl.h.