DLR-SC/tigl #890: Unexpected results from CTiglInterpolatePointsWithKinks

Leitner224, 2022-03-28:
Hello.
We happened upon an issue during the generation of a fuselage with defined kinks, where the computed profile curves overshoot on a kink edge as shown in the pictures below.
[image]
[image]

We are not sure if this is a problem with our definition or if there is an issue with the loft generation.
If it is possible, could you please check our .XML file where the issue happens for us ? It is attached below.
[kinkOvershoot.zip](https://github.com/DLR-SC/tigl/files/8360854/kinkOvershoot.zip)

Best regards and thank you for your time,
Leitner A.

joergbrech, 2022-03-28:
Thanks for opening an issue! 

In CPACS all indices start counting at One. Is it possible, that you count your point indices for the kinks starting at Zero? I plotted the `FuselageProfile2` in matplotlib and marked the kinks:

[image]

If I shift all indices by +1 [kinkOvershoot.zip](https://github.com/DLR-SC/tigl/files/8361481/kinkOvershoot.zip) there are no overshoots. Is this what you were going for? 

[image]

joergbrech, 2022-04-04:
@Leitner224 could you please check if my assumption of an off-by-one error was right?

rainman110, 2022-04-04:
For me it seems, that the kinks are not placed anymore at the right position if you compare againts the profile that you draw earlier. There are these strange osscillations right before the kink. In the original image, it looked correct but it had this overshoot that probably came from reparametrization.

rainman110, 2022-04-04:
Obviously, I implemented the indices starting from 0. Unfortunately, this does not really comply to the rest of the numbering in cpacs. see here: https://dlr-sc.github.io/tigl/towards-tigl-31.html

Leitner224, 2022-04-05:
@joergbrech sorry for my late response. As rainman110 pointed out, there is no overshoot after the +1 shift, but the kinks are missplaced in this case.

rainman110, 2022-04-05:
Just a guess, but I suppose the overshooting is caused by the reparametrization. In particular, because only one profile overshoots. This is done when creating the loft from the profiles. So it is probably a good idea to look at the profiles before lofting.