DLR-SC/tigl #901: Add Ubuntu 20.04 to open build service?

ybeyer, 2022-04-22:
Hello, I am using the TiGL Matlab interface in my research project ([SE2A](https://www.tu-braunschweig.de/se2a/research/projects/area-b/b23)).
For this I work on a computer with Ubuntu 20.04.
However, if I see it correctly, there are no pre-compiled releases for Ubuntu 20.04.
I did manage to install TiGL including Matlab interface with the available documentation, but it was quite time-consuming.
Have you thought about including Ubuntu 20.04 in the Open Build Service?
I would be very happy about that.
Thank you very much and best regards.

joergbrech, 2022-04-23:
Thanks for opening an issue! Tigl is actually being built for Ubuntu 20.04 on OSB, but for some reason the packages are not being published...I wasn't aware of that and will investigate.

joergbrech, 2022-04-25:
Fixed: https://software.opensuse.org/download.html?project=science:dlr&package=tigl3

closed 2022-04-25 by joergbrech

rainman110, 2022-04-26:
@joergbrech How did you fix this? I was always annoyed, that somehow the repo was never published.

rainman110, 2022-04-26:
Ah.... Thanks for the explanation. I am wondering, why older TiGL and OCE packages blocked newer packages that don't have any dependency relationship. I would have never figured this out 🤦