DLR-SC/tigl #919: Remove reference to Python36 from Python install instructions

jnwalther, 2022-08-02:
The install instruction for TiGL in Python for the latest version (3.2.3) found [here](https://dlr-sc.github.io/tigl/doc/3.2.3/tigl_installation.html) still request an installation of Python 3.6, which has been EOL since the beginning of the year.

As far as I can see, builds are available for more recent versions e.g. 3.9, which install just fine. Therefore, unless you have a specific reason not to do it, I suggest you remove the argument `python=3.6` from the instructions.

joergbrech, 2022-08-04:
Good point. There is no special reason for the argument to be there, the installation instruction just didn't "age well".

joergbrech, 2022-10-19:
The install instruction could use a general overhaul, see also https://github.com/DLR-SC/tigl/discussions/929#discussioncomment-3902820.

We could also think about moving the build instructions to the documentation.

closed 2024-01-09 by AntonReiswich