DLR-SC/tigl #928: how to configure the header file tigl_config.h and run the program?

tigllearner, 2022-10-17:
I am the beginner of Tigl,I want to run the code in the "unittests" folder,but I find some header files such as tigl_config.h don't exist，I want to use cmake to configure that header file by tigl_config.h.in ,but I didn't success ,so I want to know if I'm doing it the right way?if not ,what should I do?thanks a lot!

joergbrech, 2022-10-17:
Hello @tigllearner, I like your user name!

If you could let me know the steps you followed to build TiGL, I might give you better pointers on what to change. 

In general, if you want to build TiGL from source, you would have to run CMake in the root directory of the TiGL repository. If you do so, CMake will automatically configure `tigl_config.h`. So I suggest creating a subdirectory `build` and invoking the `cmake` command from there, passing the tigl root directory as the argument. Note that the unit tests are not build automatically. You have to explicitly set the CMake Option `TIGL_BUILD_TESTS` to be `ON`. Also, you have to make sure that TiGL's requirements (OpenCascade and TiXI3) are installed and can be foun by CMake.

```console
mkdir build
cd build
cmake -DTIGL_BUILD_TESTS=ON ..
```

joergbrech, 2022-10-17:
I will move this issue to the discussions, if you don't mind.