DLR-SC/tigl #933: Wing Cell lofting error with multiple sections over constant cross-section

ewerthen, 2022-11-22:
Use-Case with 5 cpacs sections and 4 segments and a constant cross-section:
[analytical_example_circ_beam_loads_composite_CPACS.zip](https://github.com/DLR-SC/tigl/files/10066345/analytical_example_circ_beam_loads_composite_CPACS.zip)

Error message:
File "C:\Anaconda3\envs\predocs\lib\site-packages\tigl3\configuration.py", line 19045, in get_loft
return _configuration.CCPACSWingCell_get_loft(self)
RuntimeError: OpenCASCADE Error: BSpline curve: at least 2 poles required

joergbrech, 2022-11-22:
I can confirm this. This is a tolerance problem caused by the `positioningOuterBorder` of the cell `span03_circ01`: If you change this value from 0.75 to 0.749999 it works. 

The problem is that the outer positioning border coincides with the segment border. Due to floating point errors we have to decide wether we need to actually cut a very very slim slice away from the face when generating the cell geometry or not based on a tolerance. Aparently the tolerance we use in the cell geometry code is incompatible with a tolerance used internally by opencascade. 

I will look into it. Until then, you should have a valid workaround.

ewerthen, 2024-10-02:
Hello, in the meanwhile we have several more complex use-cases where we get this problem. Do you know when/if you could fix this bug?

ewerthen, 2024-10-08:
it seems to be not only a tolerance problem, but also a bug in the cell generation. The cells are overlayed although the outer positioning border do not coincide with the segment border. Please find a picture and the use-case below:
[image]

[IEA-22-280-RWT_DLR_loads_CPACS.zip](https://github.com/user-attachments/files/17292386/IEA-22-280-RWT_DLR_loads_CPACS.zip)

AntonReiswich, 2024-10-09:
Hello @ewerthen, thank you for the insight. We will discuss this and get back to you regarding the bug in the beginning of next week again.

svengoldberg, 2024-10-14:
Hello @ewerthen! We made some investigations regarding your issue. Could you please checkout the branch belonging to the PR above ([issue_cells_overlap](https://github.com/DLR-SC/tigl/tree/issue_cells_overlap)) and test whether it works for you?
In our eyes, the problem is fixed and the cells do not overlap anymore. Nevertheless, we probably want to change the procedure in the future to work differently

ewerthen, 2024-10-14:
Hi, thank you very much! For testing we would need a compiled version for python. Could you provide us a python wheel?

rainman110, 2024-10-15:
Just replace the tigl3.dll in your conda installation with the one from the CI build: https://github.com/DLR-SC/tigl/actions/runs/11325889409/artifacts/2052673813

You don't need to update the python bindings, since they use dynamic linking.

ewerthen, 2024-10-15:
It doesn't seem to work for us:
[image]

joergbrech, 2024-10-17:
Should be fixed with #1026. Note that this issue results from tolerance issues when trying to trim a face very close to one of its surrounding edges. The Fix in #1026 reduces the tolerance that was defined earlier but doesn't solve the issue in principle. This would require a large rework of the cell code (for example "growing" the faces before trimming )

closed 2024-10-17 by joergbrech