DLR-SC/tigl #955: Support CPACS 3.4

marcengelmann, 2023-03-28:
Is there a timeframe or some workaround to support cpacs 3.4 files? The read-in crashes due to the new header definition in the cpacs files.

MarAlder, 2023-12-02:
This should not happen, @marcengelmann, TiGL supports all CPACS v3.x versions. If `cpacs/header/cpacsVersion` is set, the expected behavior for CPACS v3.4 and v3.5 (coming soon) is a warning `CPACS dataset version is higher than TIGL library version!`. 

Note that the `cpacsVersion` element is still available in CPACS v3.5 (but marked as deprecated, see [here](http://dlr-sl.github.io/CPACS/html/a430c658-6111-6868-d855-201b284ca81e.htm)), so TiGL should open the files correctly. In the future, we will provide a TiGL patch to support the new `versionInfos` node as well. The deprecated node will then be removed from future CPACS releases.

marcengelmann, 2023-12-13:
Hi Marko, thanks for the reply. I took a look again and somehow this deprecated cpacsVersion node is missing in the 3.4 schema file we have been using at BHL. But on the latest schema version available online this node is there, so everything is fine.

closed 2023-12-13 by marcengelmann