DLR-SC/tigl #966: Installation of TIGL-Viewer under Linux with Conda

MarkusZimmerDLR, 2023-05-05:
Dear DLR-SC-TIGL-Team,

we are trying to install the TIGL-Viewer under RedHat8 (Linux Distro) with Conda:

```
conda create -n tigl_viewer_test
conda activate tigl_viewer_test
conda install tigl3 -c dlr-sc
```
Everything seems to be compiled correctly. However, we cannot find a TIGL-Viewer-binary anywhere.
Is the viewer no longer shipped together with the TIGL-Library in the conda-packages?

Cheers,
Markus

rainman110, 2023-05-05:
Tiglviewer was never built in the tigl conda packages as far as I remember. At least not on Linux.

MarkusZimmerDLR, 2023-05-05:
We had a running TIGL-Viewer on an older system. So I simply assumed it to be this way.
What is the current standard way of installing TIGL-Viewer on Linux?

joergbrech, 2023-05-05:
See https://github.com/DLR-SC/tigl-conda/issues/15. It is planned to add the TiGL-Viewer to the conda packages.

MarkusZimmerDLR, 2023-05-05:
The Issue is from Sep 2020...

joergbrech, 2023-05-05:
> What is the current standard way of installing TIGL-Viewer on Linux?

https://dlr-sc.github.io/tigl/doc/latest/tigl_installation.html. Unfortunately, we don't have current TiGL Releases for recent Redhat versions, see https://github.com/DLR-SC/tigl/issues/895. I hope that we can change this in the near future. In the meantime, I suggest we add TiGLViewer to the Anaconda packages.

closed 2025-09-24 by joergbrech