DLR-SC/tigl #969: visualize tank geometry in tigl, apply fuselage methods for it

sfreund-DLR, 2023-09-08:
Hi,
with the new [cpacs tank definition](https://github.com/DLR-SL/CPACS/issues/786) finalized, I'd like to request tigl support for it. The tank geometry is defined alike the fusealge geometry starting at the "hulls"-element. These would be the requested items

- visualize the tank in tigl viewer
- provide the tigl-fuselage alike methods for the tank

An examplary cpacs dataset is under development and will be added when finished.

This is an examplary view of an aircraft with tanks 
[image]

MarAlder, 2024-03-20:
[image]

The implementation of the tanks in TiGL is nearing completion (the figure above shows the TiGL implementation of a highly simplified tank with two layers (green) in a fuselage (blue)). I am still working on the transfer of some fuselage operations. @sfreund-DLR, would you provide a (Python) prototype for calculating the geometry according to the simplified parameters, ideally with pythonOCC? 

Note that I moved the `transfomation` element one level higher.
[image]

MarAlder, 2024-04-05:
Linking the Python prototype for `designParmeters`: [tanks.zip](https://github.com/DLR-SC/tigl/files/14881519/tanks.zip)

MarAlder, 2024-05-14:
I propose to restructure the schema so that we can define the `designParameters` as an alternative to `segments` and `sections`. This makes it possible to define multiple `hulls` via the simplified approach. Furthermore, all other elements (`structure`, `cutOuts`, maybe buffels/walls) can be used in the same way for both approaches. The schema also looks cleaner. Disadvantage: The order of the elements must be adhered to (choice elements unfortunately force `xsd:sequence`).

[image]

What do you think, @sfreund-DLR?

sfreund-DLR, 2024-05-16:
good idea!

MarAlder, 2024-05-24:
TiGL philosophy question: The third ellipsoid dome in this example has a negative half-axis fraction. Should we set the values to the next reasonalbe values (values <0 to =0; values >1 to =1), throw a warning, and build the geometry, or throw an error and not build the geometry?

Currently I implemented the first approach:
[image]
[image]

```xml
                    <hull uID="genericTank3_ellipsoidDome2">
                      <name>Hull with ellipsoid dome</name>
                      <description>This hull has an invalid halfAxisFraction. It is used for testing of exeption handling.</description>
                      <transformation>
                        <translation>
                          <x>1</x>
                        </translation>
                        <scaling>
                          <x>1</x>
                          <y>0.7</y>
                          <z>0.7</z>
                        </scaling>
                      </transformation>
                      <cylinderRadius>1.0</cylinderRadius>
                      <cylinderLength>2.0</cylinderLength>
                      <domeType>
                        <ellipsoid>
                          <!-- This should throw a warning and be set to 0 -->
                          <halfAxisFraction>-0.2</halfAxisFraction>
                        </ellipsoid>
                      </domeType>
                    </hull>
```

MarAlder, 2024-05-30:
Thanks for your feedback, also @AntonReiswich and @sdeinert via mail. I think I'll try to avoid the implicit assumptions and handle meaningless parameters with `CTiglError`. Especially in automated design space exploration, users might miss some warnings. The TiGL viewer will not crash and the Python wrapper will throw a clear exception message. Somehow I feel better about this. But still open for discussion.

Another point: Yesterday we discussed that some edge cases might be difficult with the idea of implementing it as a fuselage child. The tanks should also be placed inside and outside the wings, pylons and nacelles. So we decided that it would be better to pull fuelTanks out of the fuselage (keeping only the fuselage-integral tanks as implemented in CPACS 3.4 and earlier) and reference them via parentUID.  I guess no one really started using the schema, so I'll bring an update with CPACS 3.5.1 after finishing the TiGL implementation (in case we discuss further changes). @MarkusZimmerDLR, @sfreund-DLR 

[image]

[image]

closed 2025-05-21 by joergbrech