DLR-SC/tigl #972: CI is failing

joergbrech, 2023-10-18:
The CI runs into two issues:

# Pull Request workflow

```
Invalid workflow file: .github/workflows/pull_request.yml#L11
The workflow is not valid. .github/workflows/pull_request.yml (Line: 11, Col: 11): Secret GH_TOKEN is required, but not provided while calling.
```

# MacOS Tests

```
UnsatisfiableError: The following specifications were found
to be incompatible with the existing python installation in your environment:

Specifications:

  - pythonocc-core=7.4.1 -> python[version='>=3.10,<3.11.0a0|>=3.8,<3.9.0a0|>=3.9,<3.10.0a0|>=3.7,<3.8.0a0|>=3.6,<3.7.0a0']
  - tixi3=3.3.0 -> python[version='>=3.10,<3.11.0a0|>=3.9,<3.10.0a0|>=3.8,<3.9.0a0|>=3.7,<3.8.0a0|>=3.6,<3.7.0a0']

Your python: python=3.11
```

rainman110, 2023-10-18:
On macos, it seems to be simply the wrong python version installed (3.11).

joergbrech, 2023-11-21:
Fixed with pushes on master ( 😱 ):
 - https://github.com/DLR-SC/tigl/commit/4baee7358273cb99857e94084ca9bed838feab38
 - https://github.com/DLR-SC/tigl/commit/8652dd748a916acc040a125e5ad76f52829035fe

closed 2023-11-21 by joergbrech