DLR-SC/tigl #985: Cross-section of fuselages segments with guide curves not interpolated correctly.

LucasKugler, 2023-12-19:
I have noticed that, when generating fuselages with guide curves, certain sections show a strange behaviour: instead of keeping a somewhat circular or ovoid shape like the sections that are being interpolated, the cross-section becomes almost diamond shaped, i.e. the faces between the guide curves get flattened.

This happens only when there is a large difference in size between the sections, and gets stronger when the size difference increases. In turn, large size differences are not a problem if there is no guide curve. 

It would be nice if this could be fixed, as the models look much cleaner with the guide curves, and it avoids to clutter the cpacs file with dozens of sections. But right now the surfaces don't look right.

<details>
<summary>Screenshots</summary>

[image]
[image]

</details>

joergbrech, 2023-12-19:
💐  Thanks for opening your first issue! Note that this is an artifact of the Coons patch algorithm which is still used to generate CPACS geometries from curve networks. 

That being said, I don't know why the Coons patch algorithm behaves this way and there is a good chance that the Gordon surface algorithm as an alternative might resolve the issue. 

Now *that* being said examining what happens in the Coons patch algorithm and/or fully migrating to Gordon surfaces will take some time and we currently don't have the capacity for it. 

In the meantime, if you could provide a minimal reproducible example, it could help us speed up the process and I might be able to help with a workaround in python *(that is a script that extracts the curve network and re-generates the surface with the Gordon algorithm)*

LucasKugler, 2023-12-19:
Thanks for the feedback. I forgot to mention that the CPACS example file simpleAircraft exhibits this behaviour.
I have also attached an additional example.
[tigl_issue#985_example.zip](https://github.com/DLR-SC/tigl/files/13713912/tigl_issue.985_example.zip)

joergbrech, 2023-12-20:
The Gordon surface algorithm creates a "better looking" loft:

[image]

The workaround using a python script I mentioned currently only works for wings and not fuselages due to #986.

LucasKugler, 2023-12-20:
Much better indeed! That's the output I would expect 👍
So if I understand correctly, the workaround is dependent on #986 being fixed first ?

joergbrech, 2023-12-20:
Yes: The workaround I had in mind was to use the internal python bindings of tigl to extract the section and guide curve wires of the fuselage and then use them as input to the gordon surface algorithm to create the fuselage shell. The guide curve wires are stored in the class `CCPACSFuselageSegments` which is unfortunately not exposed as part of the python bindings.