DLR-SC/tigl #987: TiGLViewer crashes when trailing edge GuideCurve is missing

LucasKugler, 2023-12-20:
In some situations, TiGLViewer crashes when a wing has a guide curve defined a the leading edge, but not at the trailing edge.

I have attached a modified verison of the CPACS example simpleAircraft, where i have added guidecurves to the htp. When commenting out one of the TE guide curves from the VTP (l.1107), TiGL fails to render it but does not crash. When doing the same to the HTP (l.1206) however, there is a hard crash.

[example_issue#987.zip](https://github.com/DLR-SC/tigl/files/13725535/example_issue.987.zip)

joergbrech, 2023-12-20:
I can confirm this. Thanks for the find!

Some more info: Commenting out the VTP TE guide curve in line 1107 gives the error:

```console
ERR 12/20 10:51:59 CTiglMakeLoft.cpp:229] Could not create loft with guide curves. Error code = 2
ERR 12/20 10:51:59 CTiglWingBuilder.cpp:148] CCPACSWingBuilder: Unable to determine wing face names from wing loft.
```

joergbrech, 2023-12-20:
I think this might be the same as https://github.com/DLR-SC/tigl/issues/891, but I would have to verify.