DLR-SC/tigl #994: Tigl viewer crashes without report when xml not valid

PSeelemeyer, 2024-01-22:
When an invalid xml scheme is opened, Tigl crashes without a report. The event is for exemple triggered by this block:

                            <positioning uID="Turbine_Blade1_Sec1_Pos">
                                <name>MyPosition</name>
                                <length>1.5</length>
                                <sweepAngle/>
                                <dihedralAngle>0.0</dihedralAngle>
                                <toSectionUID>ExpTurb_Blade1_Sec1</toSectionUID>
                            </positioning>

where the sweep angle should be defined as

                                <sweepAngle>0.0</sweepAngle>

, however a report from tigl about the error would be helpful.

joergbrech, 2024-01-23:
Thanks for opening an issue here! Would you be able to upload a minimal example CPACS file here, so that I can try to reproduce this?

joergbrech, 2024-01-23:
I can confirm this. I believe the crash originates [in this line](https://github.com/DLR-SC/tigl/blob/ab2e70492bdba2274783c0f5cac80a4b248f167d/src/generated/CPACSPositioning.cpp#L133), so this may be a tixi bug.